openbmc/openbmc #616: [ Barreleye VERSION="v1.99.0-145" ] Phosphor Setting Main pid is Zero

gkeishin, 2016-09-30:
The Phosphor setting pid shouldn't be Zero but we are seeing now.  here is a comparison with older build vs latest.

Any Change made this impact ?

root@barreleye:~# cat /etc/version
20160930013645
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-145"
VERSION_ID="v1.99.0-145-gf37eede-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-145"
BUILD_ID="v1.99.0"
root@barreleye:~#

**New build test run**

14:26:33.446 	INFO 	Executing command 'systemctl -p MainPID show settings.service| cut -d = -f2'. 	
14:26:33.841 	INFO 	Command exited with return code 0. 	
14:26:33.842 	INFO 	${stdout} = 0 	
14:26:33.842 	INFO 	${stderr} = 
14:26:33.844 	INFO 	${MainPID} = 0

In older test this MainPID  is NOT Zero

**Example from old log** 

Start / End / Elapsed: 	20160928 00:02:11.224 / 20160928 00:02:11.691 / 00:00:00.467
00:02:11.225 	INFO 	Executing command 'systemctl -p MainPID show settings.service| cut -d = -f2'. 	
00:02:11.690 	INFO 	Command exited with return code 0. 	
00:02:11.690 	INFO 	${stdout} = 1031 	
00:02:11.691 	INFO 	${stderr} = 
00:02:11.692 	INFO 	${MainPID} = 1031

williamspatrick, 2016-09-30:
Is the settings service even running?  We have had some problems with it lately and maybe those are not fixed?

geissonator, 2016-09-30:
yeah, still got problems :(

```
Sep 28 14:05:08 barreleye systemd[1]: Started Phoshpor System Manager.
Sep 28 14:05:10 barreleye settings_manager.py[767]: Traceback (most recent call last):
Sep 28 14:05:10 barreleye settings_manager.py[767]:   File "/usr/sbin/settings_manager.py", line 186, in <module>
Sep 28 14:05:10 barreleye settings_manager.py[767]:     obj = HostSettingsObject(bus, OBJ_NAME, s.SETTINGS, "/var/lib/obmc/")
Sep 28 14:05:10 barreleye settings_manager.py[767]:   File "/usr/sbin/settings_manager.py", line 53, in __init__
Sep 28 14:05:10 barreleye settings_manager.py[767]:     shk['default'])
Sep 28 14:05:10 barreleye settings_manager.py[767]:   File "/usr/sbin/settings_manager.py", line 76, in set_settings_property
Sep 28 14:05:10 barreleye settings_manager.py[767]:     self.Set(DBUS_NAME, name, str(value))
Sep 28 14:05:10 barreleye settings_manager.py[767]:   File "/usr/lib/python2.7/site-packages/obmc/dbuslib/bindings.py", line 82, in Set
Sep 28 14:05:10 barreleye settings_manager.py[767]:     self.validator(interface_name, property_name, new_value)
Sep 28 14:05:10 barreleye settings_manager.py[767]:   File "/usr/sbin/settings_manager.py", line 171, in input_validator
Sep 28 14:05:10 barreleye settings_manager.py[767]:     self.validate_list_ignore_case(shk['allowed'], value)
Sep 28 14:05:10 barreleye settings_manager.py[767]:   File "/usr/sbin/settings_manager.py", line 111, in validate_list_ignore_case
Sep 28 14:05:10 barreleye settings_manager.py[767]:     raise ValueError("Invalid input. Data not in allowed values")
Sep 28 14:05:10 barreleye settings_manager.py[767]: ValueError: Invalid input. Data not in allowed values
Sep 28 14:05:11 barreleye systemd[1]: phosphor-settings.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 14:05:11 barreleye systemd[1]: phosphor-settings.service: Unit entered failed state.
Sep 28 14:05:11 barreleye systemd[1]: phosphor-settings.service: Failed with result 'exit-code'
```

williamspatrick, 2016-09-30:
This indicates that the value being restored from /var/lib/obmc (or defaulted) is invalid.

geissonator, 2016-09-30:
I added a few prints:

```
{'default': 'default', 'validation': 'list', 'type': 's', 'name': 'boot_flags', 'allowed': ['Network', 'Disk', 'Safe', 'CDROM', 'Setup', 'default']}
shk[name] = boot_flags
shk[type] = s
shk[default] = default
bmcv = 3ab56f
type = s
Invalid input is 3ab56f
```

geissonator, 2016-09-30:
"rm /var/lib/obmc/boot_flags" seemed to fix the issue, I'll keep an eye out for it to come back.

This is still 0 though, not sure what that means:
systemctl -p MainPID show settings.service| cut -d = -f2
0

ps -ef | grep setting
  678 root       0:04 {settings_manage} /usr/bin/python -u /usr/sbin/settings_manager.py

vishwabmc, 2016-10-03:
root@barreleye:~# systemctl -p MainPID show phosphor-settings.service| cut -d = -f2
704
root@barreleye:~# 

The Service file name has changed to phosphor-settings.service.