openbmc/openbmc #622: Set the mac address if if the requested value had the locally admin mask

ratagupt, 2016-10-03:
Currently network manager doesn't look  at the Mac address whether it is locally administered address or not.
We should put a validation check there.

amboar, 2016-10-04:
@ratagupt What is the use case here? It's not clear to me what the goal of this issue is. What are we validating?

williamspatrick, 2016-10-04:
@amboar - We were looking through some of the IPMI commands related to network settings and decided that we were not handling the "set mac address" type interfaces correctly.

What we would like the code (both REST and IPMI) to do is the follows:
    1. If the user sets a MAC address, it must be a local admin type MAC.
    2. If a local admin type MAC is currently set, we would allow the user to write the value located in the VPD (restore hardware value).

Do you have any problem with this limitation on arbitrary MAC addresses from a field perspective?

mdmillerii, 2016-10-04:
We  need to support systems that don't have a defined (or known to the stack) location for the assigned MAC address.

williamspatrick, 2016-10-04:
@mdmillerii - That would be a manufacturing interface on the VPD interface object, right?

geissonator, 2016-10-05:
Ratan, please break down task list for review in our next meeting.  Thanks!

amboar, 2016-10-05:
@williamspatrick that all sounds sensible to me, it just wasn't clear to me from the initial description.

mdmillerii, 2016-10-06:
@williamspatrick I am concerned that we not limit the ability to set the mac if the system does not have a VPD object with a defined container for the MAC address.

geissonator, 2016-10-13:
https://gerrit.openbmc-project.xyz/#/c/711/

closed 2016-11-08 by williamspatrick