openbmc/openbmc #625: Test P9 OCC data against OpenBMC OCC driver

geissonator, 2016-10-03:


amboar, 2016-10-04:
@geissonator can you add a description here? Is the P8 OCC driver expected to work without modification for P9? The OCC driver could do with some love regardless, it's a bit of a maze.

geissonator, 2016-10-04:
This sprint will focus on #624 so we'll move this to backlog for now.  Needs to be broken down further.

eddiejames, 2016-10-24:
Task list:

- [ ] implement the different P9 data structures in the OCC driver (different sensor formats, etc)
- [ ] mock up or obtain P9 OCC poll response data
- [ ] create quick test framework to read in and parse P9 OCC poll response data

eddiejames, 2016-11-01:
[p9_occ_results.txt](https://github.com/openbmc/openbmc/files/565042/p9_occ_results.txt)

Tested with a chunk of poll response data from Martha. Everything appears to be in order.

geissonator, 2016-11-30:
Still under merge review so moved to relevant sprint.

geissonator, 2017-01-18:
Upstream review still ongoing.

geissonator, 2017-03-24:
Pulling out of committed since we can't commit to when upstream will merge this.

closed 2017-05-10 by rfrandse