openbmc/openbmc #627: [skeleton/openbmc-v1.0-stable] Fail sync MAC from io_board fru if using random MAC address by default

williamspatrick, 2016-10-04:
@johnhcwang commented on [Tue Oct 04 2016](https://github.com/openbmc/skeleton/issues/129)

If we didn't set 'ethaddr' in uboot, random locally administered address would be used by default.
In this case, MAC can't be sync even if we had configured the address in io_board FRU Custom Filed 2 since [here](https://github.com/openbmc/skeleton/blob/openbmc-v1.0-stable/pyinventorymgr/sync_inventory_items.py#L73).

---

@gwshan commented on [Tue Oct 04 2016](https://github.com/openbmc/skeleton/issues/129#issuecomment-251335468)

The description from @johnhcwang is accurate and complete. I'm putting a note here for the correct procedure to update the MAC address manually:

- Bring down the interface      ("ifconfig eth0 down")
- Set the new MAC address   ("ifconfig eth0 hw ether xx:xx:xx:xx:xx:xx")
- Bring up the interface          ("ifconfig eth0 up")

---

@williamspatrick commented on [Tue Oct 04 2016](https://github.com/openbmc/skeleton/issues/129#issuecomment-251417184)

@gwshan - Please be careful about recommending commands like this.  We use systemd's networkd for controlling the network interfaces, so by doing these ifconfig commands you are bypassing that.

williamspatrick, 2016-10-04:
@johnhcwang - I moved this to the openbmc/openbmc repository.  That is where we prefer most issues be opened.

anoo1, 2016-10-06:
The code currently uses the mac provided by the GetHwAddress dbus interface to make the comparison, but this interface returns the mac that's set in 'org.freedesktop.network1.Link', (which includes the random mac), not what's set in u-boot.
It'll be more accurate for the sync code to look at u-boot directly. Will make that change.

closed 2016-11-21 by williamspatrick