openbmc/openbmc #628: Pin mux settings put in MRW and incorporated into device tree

geissonator, 2016-10-04:


amboar, 2016-10-05:
Is there a plan yet for how this is going to be achieved? My gut feeling is you're going to have to put some devicetree-specific information into the MRW. Is that on the cards?

spinler, 2016-10-05:
There is an XML model of the AST2500 that we can put whatever we need into it, and the MRW supports the concept of selecting pin groups differently based on the system the chip is in.

First I need to gather the requirements.   What pingroup related information needs to go into witherspoon.dts?

geissonator, 2016-10-05:
### Tasks

- [ ] Gather requirements, get data into MRW
- [ ] Write code to parse from mrw and create device tree

TBD: Need more discussion with LTC team.  Setup something Thursday night.

amboar, 2016-10-05:
@spinler For a reference, check out the [dtsi for the AST2400](https://github.com/openbmc/linux/blob/dev-4.7/arch/arm/boot/dts/aspeed-g4.dtsi#L77), it has has a pinctrl node with subnodes for all "pingroups" or mux functions (pins, groups and functions are separate concepts in the Linux pinctrl subsystem, but are obviously tied together by the platform's pinmux implementation):

```dts
...
pinctrl: pinctrl@1e6e2000 {
    compatible = "aspeed,g4-pinctrl";

    pinctrl_i2c3_default: i2c3_default {
        function = "I2C3";
        groups = "I2C3";
    };
...
```
From there, in the platform dts, [e.g. Palmetto](https://github.com/openbmc/linux/blob/dev-4.7/arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dts#L68), we reference the pinctrl subnodes to configure the desired functionality:

```dts
...
&pinctrl {
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_flbusy_default &pinctrl_flwp_default>;
...
```
In this case we're "hogging" pins by requesting them and their function inside the pinctrl node. Typically you add the pinctrl properties under the devicetree node for your device of interest. For example, function I2C3 would be requested in the devicetree node associated with the I2C3 controller.

spinler, 2016-11-03:
@amboar, do you know why the SPI1 node in Witherspoon needs this?
```
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_spi1debug_default>;
```

shenki, 2016-11-04:
The strapping of the processor is incorrect. I have let Sertac know that this needs to be fixed in future machines.

We are in code review for a software work around. Once we have settled on that, the correct node will look like this:

```
    pinctrl-names = "default";
    pinctrl-0 = <&pinctrl_spi1master_default>;
```

amboar, 2016-11-04:
@spinler further to @shenki 's response, it's telling the hardware that we want to use the associated pins for SPI. This is already the case as the configuration is held in the hardware strapping register (and so we would carry on working without it), but we need it to tell the pinctrl/pinmux subsystems about the required hardware state. pinctrl provides mutual exclusion for pins, which means that no-one else can come in and disturb the hardware (e.g. by requesting those pins as GPIO).

spinler, 2016-11-04:
@shenki @amboar Thanks for the explanation.
For the lines needed in the system level device tree, will it always look like?
```
pinctrl-names = "default";
pinctrl-0 = <&pinctrl_XXX_default>;
```
Where XXX is listed in pinctrl-aspeed.txt?  So, if I needed to dynamically generate these 2 lines, I could hardcode everything but the XXX, which I would get from the chip specific MRW data?  The XML model has an element representing this SPI1 interface (called a SPI master unit in MRW lingo), so inside of that I could add its pinctlr function that would be needed when this unit is enabled.

amboar, 2016-11-04:
> For the lines needed in the system level device tree, will it always look like?

> pinctrl-names = "default";
> pinctrl-0 = <&pinctrl_XXX_default>;

Yes and no. You will always need the two properties in some form, but the number of functions you'll need to request will vary. For example for spi2 as it stands requires [something like Xo's Zaius patch](http://patchwork.ozlabs.org/patch/690220/):

```
pinctrl-names = "default";
pinctrl-0 = <&pinctrl_spi2ck_default
                         &pinctrl_spi2cs0_default
                         &pinctrl_spi2cs1_default
                         &pinctrl_spi2miso_default
                         &pinctrl_spi2mosi_default>;
```

@shenki isn't happy with this specific example because spi2 requires most of those signals to properly act as a bus: They could be grouped into one function with some modifications to the driver.

UARTS are another problem similar to SPI2, except this time it's less obvious how to make a pin group that's useful to everyone, so their signals are individual functions as well.

But even if we have functions that enable all the pins for a bus, [this doesn't mean you won't encounter the problem either](http://patchwork.ozlabs.org/patch/690630/):

```diff
 	pinctrl-names = "default";
-	pinctrl-0 = <&pinctrl_rgmii1_default>;
+	pinctrl-0 = <&pinctrl_rgmii1_default &pinctrl_mdio1_default>;
 };
 
 &mac1 {
 	status = "okay";
 
 	pinctrl-names = "default";
-	pinctrl-0 = <&pinctrl_rgmii2_default>;
+	pinctrl-0 = <&pinctrl_rgmii2_default &pinctrl_mdio2_default>;
 };
```

In this case the evb has dedicated network phys and isn't using NCSI, so we must mux the MDIO functions as well. 

> Where XXX is listed in pinctrl-aspeed.txt? So, if I needed to dynamically generate these 2 lines, I could hardcode everything but the XXX, which I would get from the chip specific MRW data? 

Yes, so long as the MRW covers the issues I outlined above.

closed 2016-12-09 by geissonator