openbmc/openbmc #629: Integrate new generated device tree files into kernel

geissonator, 2016-10-04:


spinler, 2016-10-05:
- [ ] Create recipe to fetch gen_devtree.pl, run it against the XML, run the dtc against that output, and then put the output of that somewhere

amboar, 2016-10-06:
@spinler My gut feeling is that you'd then need to patch [kernel-cuimage.bbclass](https://github.com/openbmc/openbmc/blob/master/meta-phosphor/classes/kernel-cuimage.bbclass#L23) to look in the right place for your generated dtb, where-ever it ends up living.

geissonator, 2016-11-28:
Moving from 11.2 into 11.16 - blocked on hostboot support of new MRW data (coming soon)

geissonator, 2016-12-15:
Open design issues here, Matt may just check in a diff of his generated file directly to kernel until we get things resolved.

rfrandse, 2017-04-26:
Putting this in the backlog for later.

spinler, 2017-09-18:
Definitely not doing this for GA.  Moving out.

spinler, 2018-02-20:
Abandoned this feature

closed 2018-02-20 by spinler