openbmc/openbmc #630: Incorporate I2C devices into MRW and get in device tree or udev

geissonator, 2016-10-04:
Still need to determine if this is just getting into MRW for udev to handle or if we're putting in device tree.

amboar, 2016-10-05:
@geissonator The I2C devices must be described in the device tree, so that's not optional. Further, as long as the kernel is configured with CONFIG_I2C_CHARDEV I don't think we need udev help. @shenki should be able to chime in here to confirm.

williamspatrick, 2016-10-05:
@amboar , @shenki - Don't we always have the option to probe from userspace?  This is exactly what lm-sensors typically does.

bradbishop, 2016-10-05:
lm sensors uses the i2c-dev driver so it can probe.  if you can know ahead of time what is there that seems better to me (note all the warnings lm sensors gives you about probing sometimes not being safe).

I think udev still has a role here...we would still get coldplug events for the i2c devices described in the DT no?  which would then kick of a systemd unit?

shenki, 2016-10-05:
Yeah, the device drivers can be attached in userspace. You can read more here:

 https://www.kernel.org/doc/Documentation/i2c/instantiating-devices

If I were implementing this I would put them all in the device tree. This way you have all the drivers loaded before entering userspace.

bradbishop, 2016-10-05:
a side question..I'm curious what we think happens in the kernel when there are remote i2c busses over fsi links.

amboar, 2016-10-05:
Ah, possible mixup on my part. I was thinking of the I2C buses not the slave devices.

jk-ozlabs, 2016-10-05:
@bradbishop : good question. We'd need to either describe those i2c busses in the device tree, or we can implement the device-to-driver binding in userspace.

For the former option, we'd need the fsi-slave-i2c driver to be able to bind to a DT node, describing the i2c (using standard bindings there). This would best be under the correct FSI bus heirachy, so we may also want the fsi core driver to have a basic understanding of FSI layout.

williamspatrick, 2016-10-05:
Since FSI is effectively dynamic, similar to USB, I don't imagine we can fully express the device information in the device tree at build time, or even if we can, there will be hotplug triggers needed as the FSI devices come and go. 

If we need some items to be static and some to be "hotplug" this seems to require addition, implementation specific, information in the MRW so we can know which is what. Wouldn't it be simpler to have them all do one way? i.e. Through userspace.

geissonator, 2016-10-05:
Sprint Planning Notes

Only do things directly wired to BMC, do downstream and FSI in later story.
Concern with kernel compatibilities?

- [ ] Get directly wired to BMC devices into MRW
- [ ] Write code to take MRW devices and put in device tree

closed 2016-11-09 by geissonator