openbmc/openbmc #631: Pull witherspoon mrw xml into openbmc build process

geissonator, 2016-10-04:
I believe Matt is waiting on Sertac to get the final pieces checked in.

spinler, 2016-10-05:
- [ ] Submit pull request to github.com/open-power/witherspoon-xml with my XML updates, and after it is accepted:
- [ ] Put the Witherspoon XML URI & REV into witherspoon.conf

geissonator, 2016-11-16:
Blocked on serverwiz2 incompatibility issues between hostboot and openbmc.  It's being worked.

amboar, 2016-11-16:
@geissonator how are serverwiz2 issues tracked? There's only one issue open in https://github.com/open-power/serverwiz/issues and it isn't related to serverwiz2/hostboot/openbmc incompatibilities.

spinler, 2016-11-16:
It's a hostboot problem, not a serverwiz problem.   I don't know if there's an issue/defect/story, but I know it's being worked.

geissonator, 2016-11-30:
Hostboot fix is supposed to be in today, will leave this story in 16.11.16 for now to see if we can get things in by next wed.

spinler, 2016-11-30:
Yea, today there is a version of the XML that works for hostboot when made with the latest common XML.  So now I can start adding all of my stuff on top of it.

geissonator, 2016-12-08:
https://gerrit.openbmc-project.xyz/#/c/1362/

closed 2016-12-08 by geissonator