openbmc/openbmc #633: Inventory reimplementation

bradbishop, 2016-10-05:
Moving to sdbus:

- [ ] "libengine" wait for property to change to a value
- [ ] "libengine" wait for property to change value
- [ ] "libengine" wait for property to change value
- [ ] "libengine" wait for property to change value
- [ ] "libengine" wait for property to change value
- [ ] "libengine" wait for property to change value
- [ ] "libengine" wait for property to change value
https://github.com/openbmc/openbmc/issues/632

bradbishop, 2016-10-20:
'find an existing or implement  'hostboot done' signaling mechanism:

```   @dbus.service.method(
       SensorValue.IFACE_NAME,
       in_signature='v', out_signature='')
   def setValue(self, value):
       SensorValue.setValue(self, value)
       if (value == "FW Progress, Starting OS"):
           self.GotoSystemState("HOST_BOOTED")```

I'm sure there is something usable here.

bradbishop, 2016-10-20:
'find an existing or implement IPMI up/down signaling mechanism'

Just going to use a power off signal for now.  Marking task as done.

mdmillerii, 2016-10-20:
One thing to be aware of, `HOST_BOOTED` is issued by both hostboot handing off control to skiboot and skiboot handing off control to the petitboot linux kernel.

Also, should this be moved elsewhere when/if the signal for this sensor is added for the [pci reset handling](https://gerrit.openbmc-project.xyz/#/c/415/)?

bradbishop, 2016-10-20:
Thanks for the heads up @mdmillerii.

Sure.  At this point I just wanted to establish that there was in fact something usable as a first-pass attempt making this work.  If I've done it correctly..moving to a different signaling mechanism later should be easy.

bradbishop, 2016-10-20:
'Find an existing or implement a coldplug finished signaling mechanism'

systemd emits plenty of signals for this.  the trick is at least one client has to have called 'subscribe' (and that client has to stay connected).

Marking task as done.

bradbishop, 2016-11-09:
I opened issues for remaining unchecked tasks:
https://github.com/openbmc/openbmc/issues/726
https://github.com/openbmc/openbmc/issues/757

closed 2016-11-09 by bradbishop