openbmc/openbmc #634: [Barreleye VERSION="v1.99.0-152" ] Inventory ERROR (name_properties) and Partition 'BMC_INV' not found

gkeishin, 2016-10-05:
root@barreleye:~# cat /etc/version
20161005002104
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-152"
VERSION_ID="v1.99.0-152-g613641e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-152"
BUILD_ID="v1.99.0"
root@barreleye:~#

root@barreleye:~# uptime
 21:39:03 up 17 min,  load average: 0.37, 1.01, 0.74
root@barreleye:~# date
Tue Sep 20 21:39:04 UTC 2016
root@barreleye:~#

**From Syslog**
``` Sep 20 21:35:35 barreleye inventory_upload.py[901]: Partition 'BMC_INV' not found !  ```

``` Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/hdd_backplane is incorrect ("BARRELEYE Exander" instead of base node name) ```

Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf0 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf3 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm4 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm5 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf4 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf7 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/bios is incorrect ("OpenPOWER Firmware" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf6 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm17 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm9 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf2 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm29 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm28 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm25 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm24 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm21 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm20 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/cpu0 is incorrect ("PROCESSOR MODULE" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/cpu1 is incorrect ("PROCESSOR MODULE" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/dimm16 is incorrect ("0x0b" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/misc is incorrect ("BARRELEYE MB" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf5 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/membuf1 is incorrect ("0000000000000000" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR (name_properties): "name" property in /bmc/inventory/io_board is incorrect ("BARRELEYE IO Board" instead of base node name)
Sep 20 21:35:35 barreleye inventory_upload.py[901]: ERROR: Input tree has errors, aborting (use -f to force output)

williamspatrick, 2016-10-05:
BMC_INV not found is not an issue. Brad can comment on the others.

bradbishop, 2016-10-10:
The other problems are covered here:

https://github.com/openbmc/openbmc/issues/554

closed 2016-10-10 by williamspatrick