openbmc/openbmc #635: [Barreleye VERSION="v1.99.0-152" ] DNS ZeroConf  avahi-daemon.service startup failure

gkeishin, 2016-10-05:
root@barreleye:~# cat /etc/version
20161005002104
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-152"
VERSION_ID="v1.99.0-152-g613641e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-152"
BUILD_ID="v1.99.0"
root@barreleye:~# uptime
 21:41:40 up 20 min,  load average: 1.71, 0.97, 0.62
root@barreleye:~# date
Tue Sep 20 21:41:42 UTC 2016

Sep 20 21:22:10 barreleye systemd[1]: avahi-daemon.service: Main process exited, code=exited, status=255/n/a
Sep 20 21:22:11 barreleye systemd[1]: avahi-daemon.service: Unit entered failed state.
Sep 20 21:22:11 barreleye systemd[1]: avahi-daemon.service: Failed with result 'exit-code'.
Sep 20 21:22:08 barreleye avahi-daemon[741]: Failed to find user 'avahi'.

shenki, 2016-10-06:
I also see this on palmetto. Note that avahi is attempting to start before the network is up.

```
[FAILED] Failed to start Avahi mDNS/DNS-SD Stack.
See 'systemctl status avahi-daemon.service' for details.
[  OK  ] Started Clear one time boot overrides.
[  OK  ] Found device /dev/ttyS4.
[  OK  ] Found device /dev/ttyVUART0.
[  OK  ] Found device /dev/ttyS0.
[  OK  ] Started Phosphor BMC0 Control.
[  OK  ] Started Phosphor Watchdog0.
[  OK  ] Started Phosphor Console Muxer listening on device /dev/ttyVUART0.
[  OK  ] Stopped Serial Getty on ttyS0.
         Starting Phosphor Inband IPMI...
[  OK  ] Started Serial Getty on ttyS4.
[  OK  ] Reached target Login Prompts.
[  OK  ] Started Phosphor Inband IPMI.
[  OK  ] Started Startup Hacks.
[  OK  ] Started Phosphor Host0 Checkstop Monitor.
[  OK  ] Started Network Service.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Network.
```

```
# systemctl status avahi-daemon.service
[[0;1;31m●[[0m avahi-daemon.service - Avahi mDNS/DNS-SD Stack
   Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; vendor preset: enabled)
   Active: [[0;1;31mfailed[[0m (Result: exit-code) since Thu 2016-10-06 06:23:19 UTC; 45s ago
  Process: 676 ExecStart=/usr/sbin/avahi-daemon -s [[0;1;31m(code=exited, status=255)[[0m
 Main PID: 676 (code=exited, status=255)

Oct 06 06:23:13 palmetto systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Oct 06 06:23:19 palmetto systemd[1]: [[0;1;39mavahi-daemon.service: Main process exited, code=exited, status=255/n/a
[[0m
Oct 06 06:23:19 palmetto systemd[1]: [[0;1;31mFailed to start Avahi mDNS/DNS-SD Stack.[[0m
Oct 06 06:23:19 palmetto systemd[1]: [[0;1;39mavahi-daemon.service: Unit entered failed state.[[0m
Oct 06 06:23:19 palmetto systemd[1]: [[0;1;39mavahi-daemon.service: Failed with result 'exit-code'.[[0m
```

ratagupt, 2016-10-06:
I was running the same command which is there in the avahi bb file.Seems  our busy box adduser doesn't support the --user-group..
root@barreleye:~# adduser --system --home /var/run/avahi-daemon --no-create-home --shell /bin/false --user-group avahi
adduser: unrecognized option '--user-group'
BusyBox v1.24.1 (2016-08-26 01:40:36 UTC) multi-call binary.

ratagupt, 2016-10-06:
I built a image with the qemu...It is working fine and seems that there is a kernel level difference between qemu and the hardware
qemu:-
======
uname -a 
Linux qemuarm 4.4.3-yocto-standard #1 PREEMPT Thu Oct 6 05:23:17 CDT 2016 armv5tejl GNU/Linux

Hardware
=========
uname -a
Linux barreleye 4.7.3-be2001133835be80ba4657232b823ba2daaac0d6 #1 Wed Sep 14 12:04:41 UTC 2016 armv5tejl GNU/Linux.
it appears that adduser command for busybox v1.24.1 doesn't support the --user-group option.

amboar, 2016-10-06:
@ratagupt are you confident the kernel configuration is the _only_ difference between your qemu and hardware tests?

williamspatrick, 2016-10-07:
@amboar - We discussed it today and then @ratagupt didn't update the issue.  kernel has nothing to do with this.  He is investigating further.

ratagupt, 2016-10-10:
Seems it is a build time issue,Some how the file which we generate as build time (/etc/passwd and /etc/group) is not correctly picked up by the image...I am looking more why the fies are not being picked up by the image?

ratagupt, 2016-10-10:
I saw in my build env that " work/palmetto-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/etc/passwd"  as well as "sysroots/palmetto/etc/passwd" both are having the avahi user...

ratagupt, 2016-10-13:
I verified that in the flash  RO is having the correct passwd file but rw is pointing to the incorrect one.Seems it is a flash update issue..

ratagupt, 2016-10-14:
We verified today that after doing a full code update this issues goes away as in the normal code update persistent files doesn't get update....

mdmillerii, 2016-10-14:
@ratagupt contacted me on a chat medium and pointed out the RO file system `/etc/passwd` was updated with an additional user but the overlay in the RW file system did not see this.

Since `/etc/passwd` is in the preserved files whitelist, updates to the base file are not updated in a normal update.  Only if the `clear_persistent_files` option is selected when the `rwfs` is supplied (either via a full `bmc` image or a `rwfs` image) will the files be copied back from the root.

@williamspatrick @bradbishop It seems we need to investigate `systemd-sysusers` as described in [stateless on pid eins](http://0pointer.net/blog/projects/stateless.html) to ensure the required users are in the password file.

gkeishin, 2016-10-18:
Tried with clearing the persistent but stills the problem.. 

[  OK  ] Reached target Login Prompts.
[  OK  ] Started Startup Hacks.
[FAILED] Failed to start Avahi mDNS/DNS-SD Stack.
See 'systemctl status avahi-daemon.service' for details.
[  OK  ] Started Network Service.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Network.
[  OK  ] Started Phosphor Inband IPMI.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login: ftgmac100 1e660000.ethernet eth0: NCSI interface up

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

[[0;1;31m●[[0m avahi-daemon.service - Avahi mDNS/DNS-SD Stack
   Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; vendor preset: enabled)
   Active: [[0;1;31mfailed[[0m (Result: exit-code) since Tue 2016-09-20 21:22:13 UTC; 3min 7s ago
  Process: 705 ExecStart=/usr/sbin/avahi-daemon -s [[0;1;31m(code=exited, status=255)[[0m
 Main PID: 705 (code=exited, status=255)

Sep 20 21:22:08 barreleye systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Sep 20 21:22:13 barreleye systemd[1]: [[0;1;39mavahi-daemon.service: Main process exited, code=exited, status=255/n/a[[0m
Sep 20 21:22:13 barreleye systemd[1]: [[0;1;31mFailed to start Avahi mDNS/DNS-SD Stack.[[0m
Sep 20 21:22:13 barreleye systemd[1]: [[0;1;39mavahi-daemon.service: Unit entered failed state.[[0m
Sep 20 21:22:13 barreleye systemd[1]: [[0;1;39mavahi-daemon.service: Failed with result 'exit-code'.[[0m

ratagupt, 2016-10-18:
I just verified that this option didn't clear the persistent file...

ratagupt, 2016-10-18:
root@barreleye:~# ls -la /etc/passwd
-rw-r--r--    1 root     root          1030 Feb 29  2016 /etc/passwd
root@barreleye:~# ls -la /etc/group 
-rw-r--r--    1 root     root           670 Feb 29  2016 /etc/group

geissonator, 2016-10-31:
Where we at with this one @ratagupt?  Have you found a good way to clear this out?

geissonator, 2016-11-10:
No updates, moving to next sprint.

sivassrr, 2016-11-14:
How about if we try to do using pflash?

geissonator, 2016-11-14:
Per sprint planning today, moving this one into the code update refactor epic.  We'll tackle once we get there.  For now, the workaround is to just do a full external write of the flash chip.

ratagupt, 2017-05-29:
@geissonator Can you please reassign to some one else?

geissonator, 2017-05-30:
Code update related so moving over to @anoo1

gkeishin, 2017-09-19:
Can we close this one @anoo1 ?

mdmillerii, 2017-09-19:
I don't think we have done anything to address the root cause.
Maybe combine it with the support user accounts issue?

gkeishin, 2017-09-21:
Which one github# would that be @mdmillerii  ? so that I can dupe on it.

mdmillerii, 2017-09-21:
#1714 is probably the closest, but it is far along doesn't really address it. 
It is the first time we need `/etc/passwd` to be persisted and will surface this issue.

 #431 is how I found it, that also doesn't apply.

The root cause here is we need to handle merging login sources as I [mentioned above](https://github.com/openbmc/openbmc/issues/635#issuecomment-253912122).

anoo1, 2017-09-21:
Moving to user account epic https://github.com/openbmc/openbmc/issues/273

rfrandse, 2018-08-17:
moved to https://github.com/ibm-openbmc/dev/issues/89

closed 2019-04-26 by stale[bot]