openbmc/openbmc #636: Firestone and Garrison failing to build

geissonator, 2016-10-05:
Looks like it started with this commit - https://github.com/openbmc/openbmc/commit/205bfb879aae9a04127a78fc8e960e24c9f3fdc4

# Link to build
https://openpower.xyz/job/openbmc-build/279/

# Error msg
```
11:22:01 ERROR: obmc-op-control-power-1.0-r1 do_patch: Command Error: 'quilt --quiltrc /var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/firestone/openbmc/build/tmp/sysroots/x86_64-linux/etc/quiltrc push' exited with 1  Output:
11:22:01 Applying patch 0001-Add-power-control-support-for-the-Garrison-system.patch
11:22:01 patching file power_control_obj.c
11:22:01 Hunk #1 FAILED at 17.
11:22:01 Hunk #2 FAILED at 67.
11:22:01 Hunk #3 FAILED at 138.
11:22:01 Hunk #4 FAILED at 230.
11:22:01 4 out of 4 hunks FAILED -- rejects in file power_control_obj.c
11:22:01 Patch 0001-Add-power-control-support-for-the-Garrison-system.patch does not apply (enforce with -f)
```

Similar one for garrison

GHF, 2016-10-05:
I broke it with https://gerrit.openbmc-project.xyz/#/c/606/ by removing what those patches patch.

williamspatrick, 2016-10-05:
@bradbishop - How do we get these patches out of here and put into the main repository?  Do we need a config flag in skeleton for this?

GHF, 2016-10-05:
@williamspatrick @bradbishop I should explain that the functionality of those patches was replaced by the new code. The fix here is to remove the per-machine patches altogether: https://gerrit.openbmc-project.xyz/#/c/689/

closed 2016-10-05 by williamspatrick

geissonator, 2016-10-05:
Thanks for the quick fix Xo, first time I've seen them all build successfully!  https://openpower.xyz/job/openbmc-build/