openbmc/openbmc #638: Create settingsd first-level categories dynamically based on inventory object.

williamspatrick, 2016-10-05:
Presently, the settingsd code will statically create 'host0' and 'bmc0'.  There should be a mapping between the category (host) and an inventory query (.../inventory/systemN) so that there is a 1-to-1 mapping between real inventory objects and real settingsd holders.  This will likely also require the file format to change so that a setting (/host/foo) is not simply in a single file (.../foo), but incorporates the inventory item number.  (ex. .../host0-foo).

geissonator, 2016-10-05:
### Proposed task list
- [ ] Look at files and scope based on category, handle duplicate names across categories
- [ ] Define a mapping between 1st level query and inventory item (work with Brad)
- [ ] Use mapping, call mapper object to find inventory object and create 1st level objects

geissonator, 2016-10-12:
Sergey, can we cross off some of the above tasks?  Can you provide links within the tasks to the corresponding gerrit code reviews?  Thanks!

geissonator, 2016-11-08:
https://gerrit.openbmc-project.xyz/#/c/719/

closed 2016-11-08 by williamspatrick