openbmc/openbmc #639: Please bump linux-obmc

GHF, 2016-10-05:
Please bump linux-obmc to at least https://github.com/openbmc/linux/commit/31491101bab1f11b01fd880d7d34dd36d26cf8fb so that MACHINE=zaius will build.

It won't actually power on the machine without @amboar's full pinctrl bindings, but I'd like to get a master revision that builds for our team.

@williamspatrick I'd mail a commit for review but I don't know how far you want to bump the recipe.

williamspatrick, 2016-10-05:
@GHF , @shenki - I'm expecting Joel to keep this up to the level he'd like to see us using.  Joel, let me know if my expectation is wrong.

amboar, 2016-10-05:
@shenki given you've ack'ed the pinctrl patches should we apply them to dev-4.7? Or better, start cooking up a dev-4.8 with them applied?

shenki, 2016-10-05:
@amboar I was waiting for Linus to ack them before applying. Has there been any activity from him on the list?

@williamspatrick Yep, that's how I'd like to operate. I'll get on this today.

amboar, 2016-10-05:
@shenki Nothing from Linus yet, maybe he's off skiing again. Anyway, the pinctrl patches aren't the subject of the issue so I'll stop that OT discussion here.

shenki, 2016-10-06:
Please review and test: https://gerrit.openbmc-project.xyz/#/c/702/

closed 2016-10-07 by williamspatrick