openbmc/openbmc #640: Master no longer builds on ppc64le - hang on gobject-introspection build

geissonator, 2016-10-06:
Everything compiled as well as it had before with commit 08ba306 on ppc64le hardware (Ubuntu running a 16.04 docker image).

The daily build picked up 02da455 today and it's now hanging during the build process.  It gets here on all machine type builds and then hangs (I'm going on 3.5 hours so far).

```
21:13:38 0: systemd-1_229+gitAUTOINC+714c62b463-r0 do_package_qa (pid 16007)
21:13:39 [110A[JCurrently 2 running tasks (3147 of 3204):
21:14:05 [42A[JCurrently 1 running tasks (3147 of 3204):
21:14:05 [42A[JCurrently 1 running tasks (3148 of 3204):
21:14:05 [42A[JCurrently 1 running tasks (3149 of 3204):
21:14:05 [42A[JCurrently 1 running tasks (3150 of 3204):
21:14:06 [42A[JCurrently 1 running tasks (3151 of 3204):
21:14:06 [42A[JCurrently 2 running tasks (3151 of 3204):
21:14:06 [42A[JCurrently 3 running tasks (3151 of 3204):
21:14:06 0: gobject-introspection-1.46.0-r0 do_compile (pid 22691)
21:14:06 [100A[JCurrently 2 running tasks (3151 of 3204):
21:14:06 [42A[JCurrently 1 running tasks (3151 of 3204):
```

The system shows this running continuously (qemu is used by the introspection code)
qemu-arm   

So it would seem something between those 2 commits caused us to rebuild the introspection code and that is not playing well on ppc64le.

amboar, 2016-10-06:
`git bisect` will help here. Should only take 4 runs to identify the commit.

williamspatrick, 2016-10-07:
Just because 08ba306 "built" doesn't mean it was good.  We build with the sstate cache reused so it would have only rebuilt gobject-introspection if it found a need to.  It is possible that it has been "broken" for a while but was just never rebuilt.  Not sure what is causing it to rebuild lately unless the cache became invalid on this ppc64 machine.

geissonator, 2016-10-07:
Started working backwards, found that for whatever reason this is the commit that broke the ppc64le builds - https://github.com/openbmc/openbmc/commit/df7dfd46d3142289ddec9d38601e4e434dfe7d7f

adamliyi, 2016-11-09:
@geissonator , you mentioned: `ppc64le hardware (Ubuntu running a 16.04 docker image)`, do you mean building in a `ppc64le machine installed with ubuntu-16.04`, or building in a `pcc64le ubuntu-16.04 docker image`?

mine260309, 2016-11-09:
I tried to build OpenBMC on Power today and meet the same error, so I believe this issue is for error when building OpenBMC on ppc64le machine, which runs ubuntu-16.04 in docker.

adamliyi, 2016-11-09:
@mine260309 , I am confused - why on Power we need another docker image to build openbmc?  I never build openbmc on Power.  By the way, could you help to let @williamli80 know how to reproduce the bug?

mine260309, 2016-11-09:
@williamli80 I think this issue can be reproduced on any Power server running Ubuntu16.04 (14.04 may reproduce as well).
Note: if the build fails with message like `ERROR: Uninative selected but not configured correctly, please set UNINATIVE_CHECKSUM[ppc64le]`, it is a known issue, and can be workarounded with removing below lines from file `yocto-poky/meta-poky/conf/distro/poky.conf`:
```
require conf/distro/include/yocto-uninative.inc
INHERIT += "uninative"  
```

geissonator, 2016-11-09:
Correct, I recreated this issue outside of docker.  It's just that we use docker to do our automated builds so it's where the issue first popped up.  This is not the uninative issue, that workaround is in place.  This is a hang in QEMU when trying to process the gobject-introspection-1.46.0-r0 package (the compile of this package uses QEMU for something).

williamli80, 2016-11-11:
Thanks to all!  I'm building it on the ppc64le now.

williamli80, 2016-11-18:
@amboar @williamspatrick @bradbishop @geissonator Andrew help to find out the useful information about this issue: 
https://lists.yoctoproject.org/pipermail/yocto/2016-April/029582.html. 
As a known issue, the introspection data cannot be built on ppc64. 
To work around, I disable the use of qemu when we build openbmc. by modifying the configuration file:
```
Add DISTRO_FEATURES_BACKFILL_CONSIDERED = "gobject-introspection-data" to build/conf/local.conf.
```
I verified on our ppc64le machine.  It can work. The build complete, and the image can boot the system.
So I think we can work around as above, and use CI test to find out if any dependency issue.

williamli80, 2016-11-22:
PR: https://gerrit.openbmc-project.xyz/#/c/1276/

williamli80, 2016-12-02:
Update: https://gerrit.openbmc-project.xyz/#/c/1276/

closed 2016-12-12 by williamspatrick

causten, 2017-01-04:
I'm not sure this is fixed but I could be doing something wrong.  I have a clean install of Ubuntu 16.04.  I cloned openbmc and only installed the packages list on openbmc/openbmc .  I still receive  the error....

ERROR: Uninative selected but not configured correctly, please set UNINATIVE_CHECKSUM[ppc64le]

causten, 2017-01-04:
Ok, I figured it out, I still needed to comment out the lines from poky.conf

yocto-poky/meta-poky/conf/distro/poky.conf

require conf/distro/include/yocto-uninative.inc
INHERIT += "uninative"