openbmc/openbmc #643: settings_manager KeyError: 'org.openbmc.settings.Host' when booting host

shenki, 2016-10-06:
I netbooted openbmc on a Palmetto with the host already powered on. I then shutdown the host (shutdown -h now on the host), and attempted to power back on with `obmcutil poweron`. There was no hostboot output, but I did notice systemd was using 100% CPU on the BMC.

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-162"
VERSION_ID="v1.99.0-162-gd4fb9ac-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-162"
BUILD_ID="v1.99.0"
```
```
Linux palmetto 4.7.6-ae3a462415c7c05a138658ef8ffa2f321262c4ec #1 Thu Oct 6 05:34:27 UTC 2016 armv5tejl GNU/Linux
```

The output of `journalctl -f` showed:

```
Oct 06 07:13:05 palmetto settings_manager.py[958]: Traceback (most recent call last):
Oct 06 07:13:05 palmetto settings_manager.py[958]:   File "/usr/sbin/settings_manager.py", line 186, in <module>
Oct 06 07:13:05 palmetto settings_manager.py[958]:     obj = HostSettingsObject(bus, OBJ_NAME, s.SETTINGS, "/var/lib/obmc/")
Oct 06 07:13:05 palmetto settings_manager.py[958]:   File "/usr/sbin/settings_manager.py", line 49, in __init__
Oct 06 07:13:05 palmetto settings_manager.py[958]:     for i in settings[DBUS_NAME].iterkeys():
Oct 06 07:13:05 palmetto settings_manager.py[958]: KeyError: 'org.openbmc.settings.Host'
Oct 06 07:13:05 palmetto systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Oct 06 07:13:05 palmetto systemd[1]: Failed to start Phosphor Settings Manager.
Oct 06 07:13:05 palmetto systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Oct 06 07:13:05 palmetto systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
Oct 06 07:13:05 palmetto systemd[1]: org.openbmc.settings.Host.service: Service hold-off time over, scheduling restart.
Oct 06 07:13:05 palmetto systemd[1]: Stopped Phosphor Settings Manager.
Oct 06 07:13:05 palmetto systemd[1]: Starting Phosphor Settings Manager...
Oct 06 07:13:08 palmetto settings_manager.py[960]: Traceback (most recent call last):
Oct 06 07:13:08 palmetto settings_manager.py[960]:   File "/usr/sbin/settings_manager.py", line 186, in <module>
Oct 06 07:13:08 palmetto settings_manager.py[960]:     obj = HostSettingsObject(bus, OBJ_NAME, s.SETTINGS, "/var/lib/obmc/")
Oct 06 07:13:08 palmetto settings_manager.py[960]:   File "/usr/sbin/settings_manager.py", line 49, in __init__
Oct 06 07:13:08 palmetto settings_manager.py[960]:     for i in settings[DBUS_NAME].iterkeys():
Oct 06 07:13:08 palmetto settings_manager.py[960]: KeyError: 'org.openbmc.settings.Host'
Oct 06 07:13:08 palmetto systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Oct 06 07:13:08 palmetto systemd[1]: Failed to start Phosphor Settings Manager.
Oct 06 07:13:08 palmetto systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Oct 06 07:13:08 palmetto systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
Oct 06 07:13:08 palmetto systemd[1]: org.openbmc.settings.Host.service: Service hold-off time over, scheduling restart.
Oct 06 07:13:08 palmetto systemd[1]: Stopped Phosphor Settings Manager.
Oct 06 07:13:08 palmetto systemd[1]: Starting Phosphor Settings Manager...
Oct 06 07:13:11 palmetto settings_manager.py[962]: Traceback (most recent call last):
Oct 06 07:13:11 palmetto settings_manager.py[962]:   File "/usr/sbin/settings_manager.py", line 186, in <module>
Oct 06 07:13:11 palmetto settings_manager.py[962]:     obj = HostSettingsObject(bus, OBJ_NAME, s.SETTINGS, "/var/lib/obmc/")
Oct 06 07:13:11 palmetto settings_manager.py[962]:   File "/usr/sbin/settings_manager.py", line 49, in __init__
Oct 06 07:13:11 palmetto settings_manager.py[962]:     for i in settings[DBUS_NAME].iterkeys():
Oct 06 07:13:11 palmetto settings_manager.py[962]: KeyError: 'org.openbmc.settings.Host'
Oct 06 07:13:11 palmetto systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Oct 06 07:13:11 palmetto systemd[1]: Failed to start Phosphor Settings Manager.
Oct 06 07:13:11 palmetto systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Oct 06 07:13:11 palmetto systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
Oct 06 07:13:11 palmetto systemd[1]: org.openbmc.settings.Host.service: Service hold-off time over, scheduling restart.
Oct 06 07:13:11 palmetto systemd[1]: Stopped Phosphor Settings Manager.
Oct 06 07:13:11 palmetto systemd[1]: Starting Phosphor Settings Manager...
```

ratagupt, 2016-10-12:
Oct 10 09:31:58 palmetto settings_manager.py[1302]: use_dhcp_ntp
Oct 10 09:31:58 palmetto settings_manager.py[1302]: Traceback (most recent call last):
Oct 10 09:31:58 palmetto settings_manager.py[1302]:   File "/usr/sbin/settings_manager.py", line 232, in <module>
Oct 10 09:31:58 palmetto settings_manager.py[1302]:     allobjects = create_object(s.SETTINGS)
Oct 10 09:31:58 palmetto settings_manager.py[1302]:   File "/usr/sbin/settings_manager.py", line 65, in create_object
Oct 10 09:31:58 palmetto settings_manager.py[1302]:     if setting['subtree']:
Oct 10 09:31:58 palmetto settings_manager.py[1302]: KeyError: 'subtree'
Oct 10 09:31:58 palmetto systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Oct 10 09:31:58 palmetto systemd[1]: Failed to start Phosphor Settings Manager.

gtmills, 2018-08-21:
@shenki This is an old build. Still an issue or can this be closed?

closed 2018-08-21 by shenki