openbmc/openbmc #644: [Barreleye VERSION="v1.99.0-161"] IPMItool command to set network override setting fails

rahulmah, 2016-10-06:
IPMItool command to set network override setting in openbmc fails due to Invalid input(Data not in allowed values). Checked journal logs and found that settings_manager.py expect "addr_type" as "STATIC", "DYNAMIC" while network override seting's ipmi raw command is designed to provide 0x00 0x00 (for dhcp) or 0x00 0x01 (for static ip).

**IPMI command executed**
'''
root@barreleye:/tmp# /tmp/ipmitool -I dbus raw 0x00 0x08 0x61 0x80 0x21 0x70 0x62 0x21 0x00 0x01 0x06 0x04 0xf4 0x52 0x14 0xf3 0x11 0xdf 0x00 0x01 0x0a 0x3d 0xa1 0x42 0x10 0x0a 0x3d 0x2 0x1
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0x8 rsp=0xff): Unspecified error
root@barreleye:/tmp# /tmp/ipmitool -I dbus raw 0x00 0x08 0x61 0x80 0x21 0x70 0x62 0x21 **0x00 0x01** 0x06 0x04 0xf4 0x52 0x14 0xf3 0x11 0xdf 0x00 0x01 0x0a 0x3d 0xa1 0x42 0x10 0x0a 0x3d 0x2 0x1
root@barreleye:/tmp# /tmp/ipmitool -I dbus raw 0x00 0x08 0x61 0x80 0x21 0x70 0x62 0x21 0x00 0x01 0x06 0x04 0xf4 0x52 0x14 0xf3 0x11 0xdf 0x00 0x01 0x0a 0x3d 0xa1 0x42 0x10 0x0a 0x3d 0x2 0x1 -vvvvvv
Running Get PICMG Properties my_addr 0x20, transit 0, target 0
Invalid Get PICMG Properties response length 1
Running Get VSO Capabilities my_addr 0x20, transit 0, target 0
Invalid response length 1
Acquire IPMB address
Discovered IPMB address 0x0
Interface address: my_addr 0x20 transit 0:0 target 0x20:0 ipmb_target 0

RAW REQ (channel=0x0 netfn=0x0 lun=0x0 cmd=0x8 data_len=27)
RAW REQUEST (27 bytes)
 61 80 21 70 62 21 00 01 06 04 f4 52 14 f3 11 df
 00 01 0a 3d a1 42 10 0a 3d 02 01
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0x8 rsp=0xff): Unspecified error
root@barreleye:/tmp#
'''

**OpenBmc fw details**
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-161"
VERSION_ID="v1.99.0-161-ge4bfd68-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-161"
BUILD_ID="v1.99.0"
root@barreleye:~# cat /etc/version
20161006024821
root@barreleye:~#

**Journal logs**
[Journal_logs_network_override.txt](https://github.com/openbmc/openbmc/files/513436/Journal_logs_network_override.txt)

rahulmah, 2016-10-12:
@ratagupt : Can you please take this issue forward?

ratagupt, 2016-10-12:
Seems vishwa is working on that..

rahulmah, 2016-10-12:
@williamspatrick : I believe this issue can be transferred to Viswa. It needs to be fixed.

williamspatrick, 2016-10-12:
I thought I had already assigned him. Is this a duplicate?

rahulmah, 2016-10-13:
@vishwabmc : As discussed, network setting override which is not getting cleared using ipmi. Please fix this also.

rahulmah, 2016-11-21:
@vishwabmc : Also see that due to this issue even the REST is failing to set network setting override. As suggested earlier, if we try to set allowed value for addr_typ to 0 and 1 (in /usr/sbin/settings_manager.py) this issue is not seen with REST and IPMI both. When you are planning to drop fix for this issue ?

```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT -d "{\"data\": \"ipaddress=7.7.7.7,prefix=16,gateway=2.2.2.2,mac=f4:52:14:f3:11:df,addr_type=0\"}" https://xx.xx.xx.xx/org/openbmc/settings/host0/attr/network_config
{
  "data": {
    "description": "Internal Server Error",
    "exception": "DBusException('Traceback (most recent call last):\\n  File \"/usr/lib/python2.7/site-packages/dbus/service.py\", line 707, in _message_cb\\n    retval = candidate_method(self, *args, **keywords)\\n  File \"/usr/lib/python2.7/site-packages/obmc/dbuslib/bindings.py\", line 82, in Set\\n    self.validator(interface_name, property_name, new_value)\\n  File \"/usr/sbin/settings_manager.py\", line 239, in input_validator\\n    getattr(self, shk[\\'method\\'])(value)\\n  File \"/usr/sbin/settings_manager.py\", line 209, in validate_net_config\\n    self.validate_list_ignore_case(allowed, value)\\n  File \"/usr/sbin/settings_manager.py\", line 170, in validate_list_ignore_case\\n    raise ValueError(\"Invalid input. Data not in allowed values\")\\nValueError: Invalid input. Data not in allowed values\\n',)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle",
      "    return route.call(**args)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 518, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 674, in wrap",
      "    return self.to_jsonp(callback(*a, **kw))",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper",
      "    rv = callback(*a, **ka)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 609, in wrap",
      "    resp = {'data': callback(*a, **kw)}",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 572, in wrap",
      "    return callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 87, in __call__",
      "    return getattr(self, 'do_' + request.method.lower())(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 277, in do_put",
      "    properties_iface.Set(iface, prop, value)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__",
      "    **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking",
      "    message, timeout)",
      "DBusException: org.freedesktop.DBus.Python.ValueError: Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/dbus/service.py\", line 707, in _message_cb",
      "    retval = candidate_method(self, *args, **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/dbuslib/bindings.py\", line 82, in Set",
      "    self.validator(interface_name, property_name, new_value)",
      "  File \"/usr/sbin/settings_manager.py\", line 239, in input_validator",
      "    getattr(self, shk['method'])(value)",
      "  File \"/usr/sbin/settings_manager.py\", line 209, in validate_net_config",
      "    self.validate_list_ignore_case(allowed, value)",
      "  File \"/usr/sbin/settings_manager.py\", line 170, in validate_list_ignore_case",
      "    raise ValueError(\"Invalid input. Data not in allowed values\")",
      "ValueError: Invalid input. Data not in allowed values",
      ""
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"
}rahulmah@rackspace:~$
```

vishwabmc, 2017-04-25:
@ratagupt @dkodihal : I believe this will not be an issue when your stories for Netwotk-Refactor/Settings refactor get merged.

What are your plans in terms of making the changes to all affected components ?. 
* Do you expect individual owners to do it / do you do it yourself / have one person do for all ?

williamspatrick, 2017-04-25:
Ratan's network manager change does not affect the host settings at this time. I do not think he is doing anything to improve the situation with this issue.

williamspatrick, 2017-05-30:
Has this been recreated on a recent code version using the latest REST APIs and IPMI server?  We do not plan to support the old REST APIs originally tested here.

closed 2017-05-30 by williamspatrick