openbmc/openbmc #645: [Barreleye VERSION="v1.99.0-161"] Valid MAC address is not taken while setting network override setting

rahulmah, 2016-10-06:
Valid MAC address is not taken while setting network override setting using ipmitool. Checked journal log and found that regex fails for MAC address f4:52:14:f3:01:df. "Invalid input. Data does not satisfy regex". But with f4:52:14:f3:11:df MAC address it works fine.

```
ipmitool command with MAC address f4:52:14:f3:11:df - worked

root@barreleye:/tmp# /tmp/ipmitool -I dbus raw 0x00 0x08 0x61 0x80 0x21 0x70 0x62 0x21 0x00 0x01 0x06 0x04 0xf4 0x52 0x14 0xf3 0x11 0xdf 0xa1 0x42 0x10 0x0a 0x3d 0x2 0x1

ipmitool command with MAC address f4:52:14:f3:01:df - Fails
root@barreleye:/tmp# /tmp/ipmitool -I dbus raw 0x00 0x08 0x61 0x80 0x21 0x70 0x62 0x21 0x00 0x01 0x06 0x04 0xf4 0x52 0x14 0xf3 0x01 0xdf 0xa1 0x42 0x10 0x0a 0x3d 0x2 0x1
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0x8 rsp=0xff): Unspecified error
root@barreleye:/tmp#
```

**OpenBmc fw details**
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-161"
VERSION_ID="v1.99.0-161-ge4bfd68-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-161"
BUILD_ID="v1.99.0"
root@barreleye:~# cat /etc/version
20161006024821
root@barreleye:~#

**Journal logs from system**
[Journal log for MAC address issue.txt](https://github.com/openbmc/openbmc/files/513583/Journal.log.for.MAC.address.issue.txt)

rahulmah, 2016-10-12:
@williamspatrick : I guess, this issue also need to be assigned to Vishwa.

williamspatrick, 2017-05-30:
Has this been recreated recently?  There has been significant changes in this area since then.

closed 2017-05-30 by williamspatrick