openbmc/openbmc #647: Ensure power on code works with Witherspoon

geissonator, 2016-10-06:
Initially this was just going to be a GPIO but maybe it changed?  This issue is to investigate if the openbmc code will work as-is to power on Witherspoon, and if not, to make the code work.

geissonator, 2016-10-07:
Uma to verify there's a way via fsp debug box to do this for initial power on.

geissonator, 2016-10-13:
Actually, there's a gpio command line tool they run on the BMC to do the power on, so we need to have that function working.

geissonator, 2016-10-13:
This will include any workarounds required for power on.

shenki, 2016-10-13:
> pdbg is currently only in the debug tarball, we're going to bring it into the main
> image with that issue and run the appropriate scripts to get power on working.

We will want to move to the latest pdbg.

@apopple will also be making a new tagged release soon, so ideally that will be the version we use

SS-IBM, 2016-11-02:
- Recipe update: Create .bb / .bbappend to reference the script
- Write a script to toggle power on GPIO, turn VCS on and off, run SBE and 'putcfam' commands

williamspatrick, 2016-11-30:
This work is not complete so it should not be closed.

geissonator, 2016-11-30:
Moving to current sprint then since reopened.

closed 2016-12-14 by williamspatrick