openbmc/openbmc #648: Witherspoon ineventory, gpio, and sensor scrub

geissonator, 2016-10-06:
This issue is to investigate all the different system specific data structures we have in openbmc code.  

- Updates to skeleton/configs/Witherspoon.py required?
- Updates to kernel system hard codes?
- Updates to device tree?

Also, need to see how this plays in with the work @spinler is doing with MRW.  His code will be in place in time for Witherspoon bringup which should reduce the overall work here.

geissonator, 2016-10-11:
Uma flashed this on our witherspoon:

[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-167"
VERSION_ID="v1.99.0-167-g7216f42-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-167"
BUILD_ID="v1.99.0"

Here are some suspicious things to look at:

### GPIO

```
Sep 28 12:19:41 witherspoon systemd[1]: Started Wait for /org/openbmc/control/power0.
Sep 28 12:19:41 witherspoon systemd[1]: Reached target Restore Last State.
Sep 28 12:19:41 witherspoon systemd[1]: Starting Restore Last State...
Sep 28 12:19:41 witherspoon system_manager.py[918]: New object: /org/openbmc/control/host0 (:1.33)
Sep 28 12:19:41 witherspoon button_reset.exe[955]: GPIO Lookup:  RESET_BUTTON = 393,both
Sep 28 12:19:41 witherspoon kernel: export_store: invalid GPIO 393
Sep 28 12:19:41 witherspoon kernel[860]: export_store: invalid GPIO 393
Sep 28 12:19:41 witherspoon system_manager.py[918]: New object: /org/openbmc/control/power0 (:1.34)
Sep 28 12:19:41 witherspoon system_manager.py[918]: SystemManager Goto System State: BMC_READY
Sep 28 12:19:42 witherspoon system_manager.py[918]: Running System State: BMC_READY
Sep 28 12:19:42 witherspoon button_power.exe[957]: GPIO Lookup:  POWER_BUTTON = 387,both
Sep 28 12:19:42 witherspoon kernel: export_store: invalid GPIO 387
Sep 28 12:19:42 witherspoon kernel[860]: export_store: invalid GPIO 387
Sep 28 12:19:42 witherspoon systemd[1]: Started Phosphor Reset Button0.
Sep 28 12:19:42 witherspoon systemd[1]: Started Phosphor Power Button0.
Sep 28 12:19:42 witherspoon system_manager.py[918]: Power GPIO config: ['SYS_PWROK_BUFF', '', [('BMC_POWER_UP', True)], []]
Sep 28 12:19:42 witherspoon hwmon.py[927]: Starting HWMON sensors
Sep 28 12:19:42 witherspoon power_control.exe[961]: Power GPIO latch output
Sep 28 12:19:42 witherspoon power_control.exe[961]: Power GPIO power good input SYS_PWROK_BUFF
Sep 28 12:19:42 witherspoon power_control.exe[961]: Power GPIO 1 power_up outputs
Sep 28 12:19:42 witherspoon power_control.exe[961]: Power GPIO power_up[0] = BMC_POWER_UP active HIGH
Sep 28 12:19:42 witherspoon power_control.exe[961]: Power GPIO 0 reset outputs
Sep 28 12:19:42 witherspoon kernel: export_store: invalid GPIO 352
Sep 28 12:19:42 witherspoon kernel[860]: export_store: invalid GPIO 352
Sep 28 12:19:42 witherspoon control_host.exe[958]: GPIO Lookup:  FSI_DATA = 352,out
Sep 28 12:19:43 witherspoon power_control.exe[961]: GPIO Lookup:  SYS_PWROK_BUFF = 346,in
Sep 28 12:19:43 witherspoon kernel: export_store: invalid GPIO 346
Sep 28 12:19:43 witherspoon kernel[860]: export_store: invalid GPIO 346
Sep 28 12:19:43 witherspoon control_host.exe[958]: GPIO Lookup:  FSI_ENABLE = 344,out
Sep 28 12:19:43 witherspoon kernel: export_store: invalid GPIO 344
Sep 28 12:19:43 witherspoon kernel: export_store: invalid GPIO 345
Sep 28 12:19:43 witherspoon power_control.exe[961]: GPIO Lookup:  BMC_POWER_UP = 345,out
Sep 28 12:19:43 witherspoon power_control.exe[961]: ERROR PowerControl: GPIO setup (rc=2)
Sep 28 12:19:43 witherspoon kernel[860]: export_store: invalid GPIO 344
Sep 28 12:19:43 witherspoon kernel[860]: export_store: invalid GPIO 345
Sep 28 12:19:43 witherspoon systemd[1]: Started Phosphor Power0 Control.
Sep 28 12:19:43 witherspoon control_host.exe[958]: GPIO Lookup:  CRONUS_SEL = 326,out
Sep 28 12:19:43 witherspoon kernel: export_store: invalid GPIO 326
Sep 28 12:19:43 witherspoon kernel[860]: export_store: invalid GPIO 326
Sep 28 12:19:43 witherspoon system_manager.py[918]: ERROR: BMC_THROTTLE not found in GPIO config table
Sep 28 12:19:44 witherspoon kernel: export_store: invalid GPIO 455
Sep 28 12:19:44 witherspoon kernel[860]: export_store: invalid GPIO 455
Sep 28 12:19:44 witherspoon control_host.exe[958]: GPIO Lookup:  IDBTN = 455,out
Sep 28 12:19:44 witherspoon systemd[1]: Started OpenPOWER Host0 Control.
Sep 28 12:19:46 witherspoon discover_system_state.py[972]: Last System State:0Power Policy:RESTORE_LAST_STATE
Sep 28 12:19:46 witherspoon power_control.exe[961]: ERROR PowerControl: GPIO open error (gpio=SYS_PWROK_BUFF,rc=2)
```

dmesg output
```
export_store: invalid GPIO 393
export_store: invalid GPIO 387
export_store: invalid GPIO 352
export_store: invalid GPIO 346
export_store: invalid GPIO 344
export_store: invalid GPIO 345
export_store: invalid GPIO 326
export_store: invalid GPIO 455
```

### LED Management
```
Sep 28 12:19:05 witherspoon systemd[1]: Started Phosphor LED Management Daemon.
Sep 28 12:19:06 witherspoon systemd[1]: org.openbmc.control.led.service: Service hold-off time over, scheduling restart.
Sep 28 12:19:06 witherspoon systemd[1]: Stopped Phosphor LED Management Daemon.
Sep 28 12:19:06 witherspoon systemd[1]: org.openbmc.control.led.service: Start request repeated too quickly.
Sep 28 12:19:06 witherspoon systemd[1]: Failed to start Phosphor LED Management Daemon.

```

### Startup Hacks
```
Sep 28 12:18:58 witherspoon led_controller.exe[888]: No LEDs present in the system
Sep 28 12:18:58 witherspoon systemd[1]: Started Phosphor LED Management Daemon.
Sep 28 12:18:58 witherspoon startup_hacks.sh[863]: /usr/sbin/startup_hacks.sh: line 7: /sys/devices/platform/ahb/ahb:apb/1e787000.vuart/sirq: No such file or directory
Sep 28 12:18:58 witherspoon startup_hacks.sh[863]: /usr/sbin/startup_hacks.sh: line 8: /sys/devices/platform/ahb/ahb:apb/1e787000.vuart/lpc_address: No such file or directory
Sep 28 12:18:58 witherspoon systemd[1]: startup-hacks.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 12:18:58 witherspoon systemd[1]: Failed to start Startup Hacks.
Sep 28 12:18:59 witherspoon inarp[879]: updating interface: eth0, [70:e2:84:14:16:6c]
Sep 28 12:18:58 witherspoon systemd[1]: startup-hacks.service: Unit entered failed state.
Sep 28 12:18:58 witherspoon systemd[1]: startup-hacks.service: Failed with result 'exit-code'.
```

spinler, 2016-10-11:
FWIW, the FRU and sensor list in Witherspoon.py is still just a copy of Firestone and definitely needs to be fixed.

geissonator, 2016-10-11:
Looks like the main thing we need ASAP is the ability for users to turn on/off power.

gpioutil -p D1 -d out -v 1      # on
gpioutil -p D1 -d out -v 0      # off
gpioutil -p D2 -d in                # PGOOD

Sep 28 12:30:39 w5 kernel: export_store: invalid GPIO 345
Sep 28 12:30:39 w5 kernel[860]: export_store: invalid GPIO 345
Sep 28 12:30:43 w5 kernel: export_store: invalid GPIO 346
Sep 28 12:30:43 w5 kernel[860]: export_store: invalid GPIO 346
Sep 28 12:30:49 w5 kernel: export_store: invalid GPIO 345
Sep 28 12:30:49 w5 kernel[860]: export_store: invalid GPIO 345
Sep 28 12:30:53 w5 kernel: export_store: invalid GPIO 346
Sep 28 12:30:53 w5 kernel[860]: export_store: invalid GPIO 346

geissonator, 2016-10-11:
Uma verified an old image provided by @williamspatrick / @shenki  during initial Witherspoon bringup still works as expected.  @amboar some updates required with our latest changes to GPIO code on Witherspoon?

shenki, 2016-10-11:
Please review and merge this change, and then reverify on the system. Out of interest, is this a pass1 or pass2 Boxelder (Witherspoon BMC) system?

 https://gerrit.openbmc-project.xyz/#/c/731/

yadlapati, 2016-10-12:
System has pass-2 Boxelder

geissonator, 2016-10-13:
Blocked on https://github.com/openbmc/openbmc/issues/662 currently with testing this latest kernel on witherspoon.

spinler, 2016-10-14:
@williamspatrick Why did you close this? Witherspoon.py is still outdated.

williamspatrick, 2016-10-14:
Ugh. It got closed because @shenki put "fixes #648" in his commit message, which it obviously doesn't.  Reopened.

adamliyi, 2016-10-26:
For device tree:
We have noticed many i2c devices definition is missing from the Witherspoon dts file https://github.com/openbmc/linux/blob/dev-4.7/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts.
Hi @spinler , is there already a Witherspoon device tree file generated from MRW? Or we may need to fill the dts file from Witherspoon.xml by hand.

spinler, 2016-10-27:
I don't have the plumbing quite ready yet to integrate into yocto.  Really though, since there is no user space code to make use of those I2C devices at the moment, you could probably just leave the in-tree device tree as it is for now and I should be able to get the MRW generated one integrated before anyone misses them.

shenki, 2016-10-27:
@spinler can you please send us a MRW generated device tree? We want it for kernel testing now.

adamliyi, 2016-10-28:
For GPIO configuration in Witherspoon.py, there are many FIXME, e.g: https://github.com/openbmc/skeleton/blob/master/configs/Witherspoon.py#L322. I added those comments based on my understanding of Witherspoon workbook. But I am not sure.
Could anyone review the GPIO configurations?

spinler, 2016-10-31:
@shenki I need to make some updates for that new SPI device driver and then I'll send it out.

adamliyi, 2016-11-01:
An related witherspoon GPIO issue: https://github.com/openbmc/openbmc/issues/710

spinler, 2016-11-01:
I took a look at the FIXMEs in Witherspoon.py.  
My only comments are:
* BMC_THROTTLE - that was an input on Firestone, so not sure why the power on sequence was ever trying to write to it.  That signal was used to indicate a single power supply DC_GOOD loss, and actually now the BMC will get that from the OCC anyway.  Don't need it for Witherspoon. (And should probably take it out of the power-on sequence.)
* IDBTN - That GPIO is a no-connect (pulled to ground) on the backplane side of the connector.  I'll ask someone about why that is.

adamliyi, 2016-11-03:
Thanks @spinler , I will commit a patch to update Witherspoon.py GPIO setting. Still not 100% sure about some of the pin as bellow. I will also ask some HW engineer:
```
Witherspoon RESET GPIO signals:
(Please refer to Witherspoon workbook Figure 46: High level PERST generation diagram)
GPIOB7 (BMC_VS1_PERST_N) - reset PXE8725?  - (Is it the same as PEX8718_DEVICES_RESET_N on Firestone?)
GPIOA3 (BMC_CP0_PERST_ENABLE_R) - reset PCIE slots? - (Is it the same as BMC_CP0_PERST_ENABLE_R and BMC_CP1_PERST_ENABLE_R on Firestone?)

Not clear about GPIO functions:
GPIOA6 (FSI_JMFG0_PRSNT_N), What is this pin used for? (same as the "CRONUS_SEL" pin in Firestone?)
GPIOI3 (FP_PWR_BTN_N) is for power button?
GPIOJ1 (BMC_NMIBTN_IN_N) is for reset button?

Compared with Firestone:
Firestone has a "CM1_OE_R_N" (GPIOQ6) pin used while resetting. Still need it in Witherspoon?
Firestone has a "BMC_THROTTLE" (GPIOJ3) pin used in booting host CPU. Still need it in Witherspoon? (Matt: Don't need it for Witherspoon).
Firestone has a "IDBTN" (GPIOQ7) pin used in boot sequence - Witherspoon has GPIOQ7 (FP_ID_BTN_N), Still need it in Witherspoon?

Compared with Zaius:
Zaius has GPIOR1 (SOFTWARE_PGOOD) connected to UCD90160, and used it when power on - What function is that? Also need for Witherspoon? It is not used in P8 (Firestone) BMC code.
```

adamliyi, 2016-11-03:
A patch submitted to clean up GPIOs in Witherspoon.py: https://gerrit.openbmc-project.xyz/#/c/992/. Please have a review - as noted by above comments - there is still open questions for this fix.

mine260309, 2016-11-03:
@Matt I traced back the `BMC_THROTTLE` code, it was added in commit 2fd67f90, which indicates it's used to "Set default LED status, pull high BMC_Throttle", by ken.sk.lai@mail.foxconn.com.
Not sure what is the "default LED", but `BMC_THROTTLE` GPIO config does exist on Palmetto, Bareleye, Firestone and Garrison.
I'm guessing that Withersoon/Romulus may has an equivalent GPIO to set "default LED", but not sure which one.

spinler, 2016-11-03:
OK, Sertac should know for sure for Witherspoon.

mine260309, 2016-11-03:
Asked Nick on Romulus, the GPIO name is changed to `N_MODE_N` for P9, and it's still `GPIOJ3`, the same as P8. Witherspoon has the same define on this.
The difference is, on P8 this GPIO is `OUT`, but it becomes `IN` on P9, the reason is quoted:
> The BMC is too slow for what this signal actually is.  It is the loss of redundancy in the power supplies.  At least in P8 time frame.
By the time the BMC has figured that out.... it is far too late to try and throttle 
Which is why it is an input.

So in BMC code, we can still define `BMC_THROTTLE` as `GPIOJ3` for Witherspoon/Romulus, but writing it to 1 just does not take effect.

adamliyi, 2016-11-04:
@spinler , Hi Matt, the inventory definition in *Witherspoon.py* are outdated, e.g.:
```
    '<inventory_root>/system/chassis/motherboard/membuf0' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
    '<inventory_root>/system/chassis/motherboard/membuf1' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
    '<inventory_root>/system/chassis/motherboard/membuf2' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
    '<inventory_root>/system/chassis/motherboard/membuf3' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
    '<inventory_root>/system/chassis/motherboard/membuf4' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
    '<inventory_root>/system/chassis/motherboard/membuf5' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
    '<inventory_root>/system/chassis/motherboard/membuf6' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
    '<inventory_root>/system/chassis/motherboard/membuf7' : { 'fru_type' : 'MEMORY_BUFFER', 'is_fru' : False, },
```
I am not clear in the future shall we generate those definition from MRW? If so, since these inventory definitions do not block Witherspoon power-on, shall we just wait for MRW become ready?

spinler, 2016-11-04:
@williamspatrick The plan is for Yi to just fix the inventory FRU list in Witherspoon.py quick until Brad finishes his inventory rewrite, or for me to turn on generating the inventory list from the MRW?

mine260309, 2016-11-04:
`IDBTN` is an `IN` GPIO for both Withspoon and Romulus, and connected with a button to signal a LED light. If we write to 0 to this GPIO, it may cause the button not working.
So for P9, we should probably remove the code related to `BMC_THROTTLE` and `IDBTN`.

@williamspatrick @shenki Do you have any idea how to manage this case: keep using the above two GPIO for P8 and remove them for P9?

Using `#if` macro is simple and  will work but it seems not a good solution.

adamliyi, 2016-11-14:
About Witherspoon power on, @mine260309 mentioned: "can go through istep 0-4 without problems." - So I think this story does not block Witherspoon power on.  For this story, here is where I am:

1. GPIO: 
    * submmited patch: https://gerrit.openbmc-project.xyz/#/c/992/. Although no feedback from Sertac. I think it is OK for powering on Witherspoon.
    * For issue openbmc/openbmc#743. @mine260309 has plan to fix it - we have solution to handle system specific GPIO

2. Inventory:
    * Track by issue: #745. I think we don't need a quick fix since inventory does not block power on for now. If it is blocking, I can fix the inventory issue case by case.

I will investigate whether there are other system specific issues.

adamliyi, 2016-11-18:
@williamspatrick , @geissonator , could you merge patch: https://gerrit.openbmc-project.xyz/#/c/992/ and close this issue.
As discussed with Andrew,  I don't think currently any system specific setting in Witherspoon.py can block powering on Witherspoon, although some configurations are outdated, we can track them in other issues. It is very likely as IPL moving forward, there will be problem related with this story arise, but that is unseen for now - So l'd prefer close this story and open new ones in future.

geissonator, 2016-11-23:
Just need another +1 from Joel and we can merge it.  Leaving in current sprint since seems really close.

geissonator, 2016-11-28:
Not merged so moving this to 11.16 sprint.

closed 2016-12-01 by williamspatrick