openbmc/openbmc #649: Ability to write hostboot scratch registers from obmc

geissonator, 2016-10-06:
For debug and testing, we need to have an interface to write the scratch registers from the openbmc.  These are set in the kernel currently so we'll need some discussion on how best to provide a debug tool that can change them.

shenki, 2016-10-10:
I think we will write a simple kernel driver to expose the scratch registers. They can then be set from userspace with sysfs or a similar API.

sivassrr, 2016-10-10:
Good to have tool like this. Hope it will be for internal purpose only.
BMC suppose not to manipulate HW registers as per the security.
 
Or is it fine ? because we are doing what we are planning to do it from OS, right.
Want to get better understanding.
 
Sivas R. S
E-mail: sivas.srr@in.ibm.com

williamspatrick, 2016-10-10:
These are registers in the SIO logic of the BMC. Hostboot (and Skiboot) read them for configuration reasons to know certain state about the BMC. For instance, SUART vs vUART and 'memboot' mode are selected this way. There is no security concern with these registers.

mdmillerii, 2016-10-10:
I think we should consider the nvmem subsystem to export these _scratch_ registers.  While its designed to extract per-chip customization for the kernel there are ways to name the bits, specify them in the device tree, and provide user interface though sysfs including read and write.

shenki, 2016-10-26:
@amboar is taking a look at this

amboar, 2018-04-16:
Sent patches to the list: 

Cover: https://lists.ozlabs.org/pipermail/openbmc/2018-April/011361.html

Patches: http://patchwork.ozlabs.org/project/openbmc/list/?series=38550&state=%2A&archive=both

amboar, 2019-03-19:
Hostboot can no-longer access the scratch registers as SIO is disabled.

closed 2019-03-19 by amboar