openbmc/openbmc #650: Execute CFAM to boot witherspoon

geissonator, 2016-10-06:
A one liner we need in openbmc code to power on witherspoon.  It requires openFSI and basic scom/mbox support.

This is what we need to start the P9 boot process

putcfam pu 2801 80000000

geissonator, 2016-10-13:
Looks like initially we're going to use pdbg to do power on work.  This will track us moving over to openFSI once ready.

shenki, 2016-10-13:
@geissonator do we have a ticket to track that? 

Alistair is about to release a major update to pdbg that improves it's stability. We should target that version for power on.

apopple, 2016-10-13:
Hey, for information once OpenFSI is ready I plan on making an OpenFSI backend for pdbg as well. This means the same commands should continue to work with pdbg but using the OpenFSI kernel driver (which will hopefully be quicker/more reliable than bit-banging from userspace :)

geissonator, 2016-10-13:
Good stuff guys, this work is tracked in https://github.com/openbmc/openbmc/issues/647.  pdbg is currently only in the debug tarball, we're going to bring it into the main image with that issue and run the appropriate scripts to get power on working.

GHF, 2016-11-19:
@geissonator @shenki How are you folks overriding the GPIOs used by pdbg? It looks like that's all hardcoded: https://github.com/open-power/pdbg/blob/master/libpdbg/bmcfsi.c#L31-L39

I don't see machine patches against pdbg in meta-witherspoon.

shenki, 2016-11-22:
We chatted on IRC about this. The goal is to get openfsi merged into the tree, and have pdbg use that.

If we don't have that in place before the end of the week we will instead apply a patch to pdbg so it can be used in the mean time.

apopple, 2016-11-22:
@shenki You will also need a direct FSI read/write interface for OpenFSI to kick off the CFAM sequence. Chris was going add this to OpenFSI at some point, I will check on the current patch series. It should be pretty easy to add if it's not there though.

geissonator, 2017-02-13:
This was done using pdbg (thanks @apopple!).  #866 tracks us moving over directly to openFSI where applicable.

closed 2017-02-13 by geissonator