openbmc/openbmc #651: Need PMBUS Power Supply device driver

spinler, 2016-10-07:
Witherspoon has 2 PMBus compliant power supplies that require monitoring, so an I2C device driver for them is required.  I'm not sure if a driver is required for the specific power supply model or not.

Joel asked for an issue to be created to track it.

shenki, 2016-10-10:
What are the part numbers that need to be supported?

spinler, 2016-10-10:
2200W HV AC/DC Power Supply
EC Number: P17531

jkeusema, 2016-10-10:
The power supply follows standard "PMBus" command set which is more or less universal across IBM power supplies. The key items we should be able to get are fault codes. The registers of primary interest would be STATUS_WORD, STATUS_VIN, STATUS_VOUT, STATUS_IOUT, STATUS_TEMPERATURE, STATUS_MFR_SPECIFIC, STATUS_FANS_1_2, STATUS_OTHER, STATUS_INPUT. When a power supply fault occurs this information should be collected for the SEL. A CLEAR_FAULTS command should be supported to clear fault codes.

INPUT_HISTORY and INPUT_HIST_SYNC is required for power trending data collection, it must be collected periodically for customer power usage information. Other telemetry information (READ_xxxxx) would be useful but not strictly required at this time (I believe).

IDLE_MODE should be supported but we want to leave the default as 00h unless we require it to be enabled.

Youngzhong, 2016-10-11:
You need do all fault flag for recording the fault SEL

zahrens, 2017-07-14:
@eddiejames Do we have a task list?

bradbishop, 2017-07-14:
This is an epic.  I didn't think we assigned epics to sprints.  The task lists would be in the stories #1884 and *1883.

closed 2017-10-10 by rfrandse