openbmc/openbmc #652: Need fully functional MAX31785 fan controller device driver

spinler, 2016-10-07:
The fan control code will need a device driver for the MAX31785 fan control chip on Witherspoon.  I heard a partial one might exist already?

agangidi53, 2016-10-08:
This is me being curious. For Barreleye we had an issue where if BMC goes down abruptly , there was no timeout capability in the fan control chip we used. Is that fixed for witherspoon ?

spinler, 2016-10-08:
Yes, Witherspoon has a watchdog circuit that the BMC has to keep pinging or the fans will go to high speed.

shenki, 2016-10-10:
Current driver:
https://patchwork.ozlabs.org/patch/672045/

Needs to be reworked:

> At this time upstream needs some significant work; the PMBus core driver
> as it currently exists does not support the required functions.
> However, the hwmon maintainer (Gunter Roeck) has said he will work on
> extending the PMBus driver and might even get something working for this
> chip with that core driver, so there is definitely progress being made.
>  I wouldn't expect anything upstream until at least kernel 4.9 or later
> though.

msbarth, 2016-10-13:
There will need to be support within the MAX31785 driver for both v1 and v2 of this chip. It is intended to have v2 within Witherspoon, but support for both will be necessary as initial cards will have v1. v2 has an extra 2bytes of data on the tach feedback register(READ_FAN_SPEED_1). The MFR_REVISION can be used to determine which version of the MAX31785 exists.

shenki, 2016-10-18:
Upstream discussion: http://www.spinics.net/lists/linux-hwmon/msg00884.html

shenki, 2016-11-16:
We have 36d6928f9964c6022298fa8579dc2f4edcaab554 in the tree.

We need the okay that this is providing the functionality that is required.

@williamspatrick can you assign to someone in Austin to chat with the hardware guys and work out what the status is?

williamspatrick, 2016-11-17:
I suspect it is not supporting the extra two bytes mentioned above because Tim was focused just on firestone. @msbarth or @spinler can you take this investigation on?

msbarth, 2016-11-17:
@shenki I'm not familiar with were this commit is, could you tell me where its at and I'll look it over.

shenki, 2017-02-27:
This is the commit I was referring to:

https://github.com/openbmc/linux/commit/36d6928f9964c6022298fa8579dc2f4edcaab554

Does this provide the features required?

msbarth, 2017-02-27:
No, this does not support the extra two bytes for the second rotor tach speed as I mentioned earlier.
There are 2 revisions of this chip used within Witherspoon, so the MFR_REVISION(9Bh) must be used to know which revision is within the system where:

- MFR_REVISION(9Bh) = 0x3030
    - one tach feedback(2 bytes) will be set within READ_FAN_SPEED_1(90h)
    - This provides the slowest reading from all rotors of a fan in reg 90h
    ex) READ_FAN_SPEED_1 = 3f 26 00 00 (slowest rotor = 9791rpms)
- MFR_REVISION(9Bh) = 0x3040
    - two tach feedbacks(4 bytes) will be set within READ_FAN_SPEED_1(90h)
    - This still preserves that the slowest reading will be in the first 2 bytes
    - Additionally, the fast rotor tach speed is within the next 2 bytes
    ex) READ_FAN_SPEED_1 = 3f 26 93 28 (slowest rotor = 9791rpms, other fast rotor = 10387rpms)

Let me know if there are any other details you need

shenki, 2017-05-11:
For the record: we don't have a datasheet that contains this extra information.

I began the implementation of a pmbus driver for the part. The kernel pmbus framework does not yet support PWM and TACH readings, so this is will require more investigation.

I will hack the extra sensor into the existing driver.

Do we have some systems that contain the newer part?

msbarth, 2017-05-11:
I checked a couple and looks like FTC level have it. You can check that a system has it by reading the MFR_REVISION reg as noted earlier where 0x3040 has both tach feedbacks.
```
root@witherspoon:~# echo 3-0052 > /sys/bus/i2c/drivers/max31785/unbind
root@witherspoon:~# i2cget -y 3 0x52 0x9B w
0x3040
root@witherspoon:~# echo 3-0052 > /sys/bus/i2c/drivers/max31785/bind
```

amboar, 2017-06-08:
More upstream discussion:

https://lists.ozlabs.org/pipermail/openbmc/2017-June/007847.html

Followed by a patch to make the driver comply with the standard hwmon sysfs attributes:

https://lists.ozlabs.org/pipermail/openbmc/2017-June/007871.html

Overall the driver won't be accepted upstream without using the pmbus hwmon subsystem.

shenki, 2017-09-25:
Andrew is tracking this for upstream inclusion

amboar, 2017-09-25:
/me shakes his fist at upstream

amboar, 2018-04-16:
Driver is upstream, though we still have out-of-tree patches for devicetree support and communications works-arounds

closed 2018-04-16 by amboar