openbmc/openbmc #653: Need device driver for ucd90160 power sequencer chip

spinler, 2016-10-07:
Witherspoon code will need to monitor it for faults, possibly using interrupts.

shenki, 2016-10-10:
* http://www.ti.com/lit/ds/symlink/ucd90160.pdf
 * http://www.ti.com/product/UCD90160

There is drivers/hwmon/pmbus/ucd9000.c that may be applicable

mine260309, 2016-10-18:
From https://patchwork.kernel.org/patch/9291515/, it looks like current driver `drivers/hwmon/pmbus/ucd9000.c` can be used to support ucd90160.

So what I plan to do is:
1. Apply the above patch to current build;
2. Add `ucd90160` at `i2c-11` in device tree;

Then I expect the driver should just work.
@shenki could you kindly help to check what the i2c address does `ucd90160` connect to? Is it 0x64 on i2c-11?

shenki, 2016-10-18:
@mine260309 Good discovery.

I do not know what address the part is at. @spinler might be able to help us there.

shenki, 2016-10-18:
@mine260309 The patch looked good to me, so I pulled it into the kernel. See https://gerrit.openbmc-project.xyz/#/c/819/

spinler, 2016-10-18:
Yea, it is 0x64 on i2c11

mine260309, 2016-10-20:
I can confirm that with the above changes (including kernel, device tree, and kernel config), 0x64@i2c-11 does work, and then we have `/sys/class/hwmon/hwmon0` linked to `../../devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-11/i2c-11/11-0064/hwmon/hwmon0`.

@shenki currently the changes to kernel and device tree are done by a patch put under `meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-kernel/linux/linux-obmc/`.
Should this change be in the current patch, or committed to linux-obmc?

If it should be committed to linux-obmc, should this story splitted into two reviews:
1. One changes in `linux-obmc`;
2. The other for changes in `openbmc`, contains the kernel config, and the bumped linux version?

shenki, 2016-10-21:
The kernel part has landed in dev-4.7, and as of bf18e18d4d8ba9cdf608438641de7cbe69f3490a it is part of the openbmc build.

The next step is to submit a change to our device trees to enable this driver.

mine260309, 2016-10-21:
@shenki let me know when openbmc's linux version is updated, and then I will update `witherspoon.cfg` to enable `CONFIG_SENSORS_UCD9000`

shenki, 2016-10-21:
The change has been posted for review:

https://gerrit.openbmc-project.xyz/#/c/876/

You can pull that change into your branch and submit a change to enable the driver in Witherspoon.

mine260309, 2016-10-24:
`witherspoon.cfg` change has been posted for review:
https://gerrit.openbmc-project.xyz/#/c/881/

closed 2016-10-25 by williamspatrick