openbmc/openbmc #654: Need device driver for RX8900CE RTC

spinler, 2016-10-07:
For witherspoon.

shenki, 2016-10-10:
* http://www5.epsondevice.com/en/products/i2c/
 * https://support.epson.biz/td/api/doc_check.php?dl=app_RX8900CE&lang=en

Has an out of tree driver. May be less hassle to write our own.

csmart, 2016-10-13:
I'll take a crack at this.

shenki, 2016-10-18:
There is upstream support in 4.9:

https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=eff6675b5cc191308c0144dcd5dd2b49257e83d7

```
	rtc: rtc@32 {
		compatible = "epson,rx8900"
		reg = <0x32>;
		epson,vdet-disable;
		trickle-diode-disable;
	};
```

We need to pull this into dev-4.7 so we can use it on Witherspoon, as well as adding it to the kernel configuration and device tree.

@csmart, we will need to find you a different driver to hack on

csmart, 2016-10-20:
\o/

shenki, 2016-10-21:
@spinler What's the i2c address of this part?

spinler, 2016-10-24:
0x32 (7 bit)

csmart, 2016-10-26:
https://patchwork.ozlabs.org/patch/686883/

shenki, 2016-10-26:
@spinler, @csmart had some questions:

This commit adds the relevant device tree entry for the Epson RX8900CE
RTC module on the I2C bus.

> Note that there are two options that we may need to set.
> 
> If there is no connected battery:
>  epson,vdet-disable;
> 
> If there is no trickle charge diode:
>  epson,vdet-disable;
>

Do you know what the battery backup configuration is for the rtc?

csmart, 2016-10-26:
@spinler looks like the circuitry is there, but it's broken out into a connector. So if there is a battery, it appears to be offboard?

csmart, 2016-10-26:
Also, v2 https://patchwork.ozlabs.org/patch/686891/

spinler, 2016-10-26:
The  battery is on a riser card, the card that either the left or right power supply plugs into (I forget which).
There is also wiring present to use a BMC ADC pin to read the  remaining battery voltage so we can generate low battery warnings.  Not sure if the driver supports that or not?  I think that would be extra credit.

csmart, 2016-10-27:
ok, thanks @spinler

shenki, 2016-11-16:
This is in the kernel and the device tree.

closed 2016-11-16 by shenki