openbmc/openbmc #655: Kernel driver for AST2500 ADC

GHF, 2016-10-10:
Need ADC for analog inputs monitoring power rails and RTC battery.

mattconolly@fb.com mentioned a forward-port/rewrite on IRC.

kc8apf, 2017-02-17:
Pushed a dev-4.7-based tree with a complete driver that has been tested on AST2400 to https://github.com/kc8apf/linux/tree/ast2400_adc_openbmc.  Many hwmon core API backports were needed.  The driver should work for AST2500 as well but I have not made the necessary dts changes or tested it.

https://github.com/kc8apf/linux/tree/ast2400_adc_for_upstream has the same driver but based on hwmon's staging tree.  This branch only contains the device tree binding documentation and the driver itself.  dts changes depend on changes that are only in openbmc/linux.

kc8apf, 2017-02-28:
Verified driver on both AST2400 (Quanta Q71L) and AST2500 (Zaius).  Sent DT bindings and driver patches to upstream linux-hwmon@.

gtmills, 2018-05-23:
@kc8apf @GHF Still a problem or can this be closed?

kc8apf, 2018-05-24:
Can definitely be closed.  ADC driver went upstream a long time ago.

closed 2018-05-24 by gtmills