openbmc/openbmc #657: [Barreleye VERSION="v1.99.0-169"]  Discover system throws DBus.Error.ServiceUnknown

gkeishin, 2016-10-12:
root@barreleye:~# cat /etc/version
20161012032551
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-169"
VERSION_ID="v1.99.0-169-g12e3d60-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-169"
BUILD_ID="v1.99.0"
root@barreleye:~#
root@barreleye:~# uptime
 21:44:20 up 22 min,  load average: 0.86, 0.92, 1.03
root@barreleye:~# obmcutil state
 = BMC_READY
root@barreleye:~#

Sep 20 21:24:09 barreleye systemd[1]: Started OpenPOWER Host0 Control.
Sep 20 21:24:09 barreleye button_power.exe[793]: GPIO Lookup:  POWER_BUTTON = 352,both
Sep 20 21:24:09 barreleye systemd[1]: Started Phosphor Power Button0.
Sep 20 21:24:09 barreleye system_manager.py[737]: New object: /org/openbmc/control/power0 (:1.51)
Sep 20 21:24:09 barreleye system_manager.py[737]: SystemManager Goto System State: BMC_STARTING2
Sep 20 21:24:09 barreleye system_manager.py[737]: Running System State: BMC_STARTING2
Sep 20 21:24:09 barreleye systemd[1]: Started Wait for /org/openbmc/control/power0.
Sep 20 21:24:09 barreleye systemd[1]: Reached target Restore Last State.
Sep 20 21:24:09 barreleye systemd[1]: Starting Restore Last State...
Sep 20 21:24:11 barreleye system_manager.py[737]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('PCIE_RESET', False), ('USB_RESET', False)]]
Sep 20 21:24:11 barreleye power_control.exe[799]: Power GPIO latch output
Sep 20 21:24:11 barreleye power_control.exe[799]: Power GPIO power good input PGOOD
Sep 20 21:24:11 barreleye power_control.exe[799]: Power GPIO 1 power_up outputs
Sep 20 21:24:11 barreleye power_control.exe[799]: Power GPIO power_up[0] = POWER_PIN active LOW
Sep 20 21:24:11 barreleye power_control.exe[799]: Power GPIO 2 reset outputs
Sep 20 21:24:11 barreleye power_control.exe[799]: Power GPIO reset[0] = PCIE_RESET active LOW
Sep 20 21:24:11 barreleye power_control.exe[799]: Power GPIO reset[1] = USB_RESET active LOW
Sep 20 21:24:11 barreleye discover_system_state.py[825]: Traceback (most recent call last):
Sep 20 21:24:11 barreleye discover_system_state.py[825]:   File "/usr/sbin/discover_system_state.py", line 59, in <module>
Sep 20 21:24:11 barreleye discover_system_state.py[825]:     pgood = getProperty(bus, dbus_objects, 'power', 'pgood')
Sep 20 21:24:11 barreleye discover_system_state.py[825]:   File "/usr/sbin/discover_system_state.py", line 53, in getProperty
Sep 20 21:24:11 barreleye discover_system_state.py[825]:     objs[key]['bus_name'], objs[key]['object_name'], introspect=False)
Sep 20 21:24:11 barreleye discover_system_state.py[825]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
Sep 20 21:24:11 barreleye discover_system_state.py[825]:     follow_name_owner_changes=follow_name_owner_changes)
Sep 20 21:24:11 barreleye discover_system_state.py[825]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
Sep 20 21:24:11 barreleye discover_system_state.py[825]:     self._named_service = conn.activate_name_owner(bus_name)
Sep 20 21:24:11 barreleye discover_system_state.py[825]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
Sep 20 21:24:11 barreleye discover_system_state.py[825]:     self.start_service_by_name(bus_name)
Sep 20 21:24:11 barreleye discover_system_state.py[825]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
Sep 20 21:24:11 barreleye discover_system_state.py[825]:     'su', (bus_name, flags)))
Sep 20 21:24:11 barreleye discover_system_state.py[825]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Sep 20 21:24:11 barreleye discover_system_state.py[825]:     message, timeout)

``` Sep 20 21:24:11 barreleye discover_system_state.py[825]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.control.Power was not provided by any .service files ```

gkeishin, 2016-10-18:
This seems to be intermittent.. If seen in latter will re-open..

sivassrr, 2016-12-20:
dbus exception observed while checking the state of the system. 
witherspoon login: root
Password: 
root@witherspoon:~# ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
caftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
t /etc/ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
version
20161219141044
root@witherspoon:~# obmcutil state
Traceback (most recent call last):
  File "/usr/sbin/obmcutil", line 189, in <module>
    obj = bus.get_object(bus_name,obj_path)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.managers.System was not provided by any .service files
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-338"
VERSION_ID="v1.99.0-338-gd6d3ef3-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-338"
BUILD_ID="v1.99.0"
root@witherspoon:~# cat /etc/version
20161219141044

sivassrr, 2016-12-20:
[debugdbusexceptionobmcutilstate.txt](https://github.com/openbmc/openbmc/files/662930/debugdbusexceptionobmcutilstate.txt)

amboar, 2016-12-20:
@sivassrr @gkeishin Please use the triple  back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

gkeishin, 2017-03-07:
Closing this as not seen in latter builds.  Will re-open if seen.

closed 2017-03-07 by gkeishin