openbmc/openbmc #658: [Barreleye VERSION="v1.99.0-169"] obmc-console-client: Can't connect to console server: Connection refused

gkeishin, 2016-10-12:
New driver failed to start to SOL client.

root@barreleye:~# cat /etc/version
20161012032551
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-169"
VERSION_ID="v1.99.0-169-g12e3d60-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-169"
BUILD_ID="v1.99.0"
root@barreleye:~#

root@barreleye:~# obmc-console-client
obmc-console-client: Can't connect to console server: Connection refused
root@barreleye:~# logout
Connection to xx.xx.xx.xx closed.

-bash-4.1$ ssh -p 2200 root@xx.xx.xx.xx
root@xx.xx.xx.xx's password:
obmc-console-client: Can't connect to console server: Connection refused
Connection to xx.xx.xx.xx closed.
-bash-4.1$

Sep 20 21:37:49 barreleye systemd[1]: Started Phosphor Host Console SSH Per-Connection (xx.xx.xx.216:40153).
Sep 20 21:37:49 barreleye dropbear[1041]: Failed loading /etc/dropbear/dropbear_dss_host_key
Sep 20 21:37:49 barreleye dropbear[1041]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Sep 20 21:37:49 barreleye dropbear[1041]: Child connection from ::ffff:xx.xx.xx.216:40153
Sep 20 21:37:49 barreleye hwmon.py[748]: Cannot read attributes: /sys/class/hwmon/hwmon4/in2_input
Sep 20 21:37:49 barreleye hwmon.py[748]: Cannot read attributes: /sys/class/hwmon/hwmon5/in2_input
Sep 20 21:37:53 barreleye dropbear[1041]: PAM password auth succeeded for 'root' from ::ffff:xx.xx.xx.216:40153
Sep 20 21:37:53 barreleye dropbear[1041]: Exit (root): Disconnect received

williamspatrick, 2016-10-12:
This implies obmc-console-server is no longer running.

geissonator, 2016-10-12:
Is this a dup of https://github.com/openbmc/obmc-console/issues/7 ?

williamspatrick, 2016-10-12:
Unknown.  Would need to see the journal information for the obmc-console.service.

adamliyi, 2016-10-26:
This should not be a duplication of openbmc/obmc-console#7.
If `obmc-console-ssh.socket` fails to start, `obmc-console-ssh@.service` will not be started (since it listens on the socket), and there will no message: `Child connection from ::ffff:xx.xx.xx.216:40153`.

It looks like the remote connection for some reason closed.
Is this issue still existing?

sivassrr, 2016-11-08:
@gkeishin  can we recheck it with the latest driver

geissonator, 2016-11-09:
@gkeishin please confirm if this is still an issue or not in latest driver

gkeishin, 2016-11-10:
@geissonator Now i dont see this but it seems more of an intermittent issues showing up now and then..   however few driver in last couple of days is showing no sign of such failure so far..

shenki, 2016-11-10:
Given you cannot reproduce, lets close the issue. Please re-open if you see it again.

closed 2016-11-10 by shenki