openbmc/openbmc #659: [Barreleye VERSION="v1.99.0-169"]  IPv6 DHCP client ?

gkeishin, 2016-10-12:
``` I don't think the driver build has IPv6 support yet... if that is a yes, what is this V6 DHCP client doing here with timeout message ? ```

``` Should we ignore this ? ```

root@barreleye:~# cat /etc/version
20161012032551
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-169"
VERSION_ID="v1.99.0-169-g12e3d60-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-169"
BUILD_ID="v1.99.0"
root@barreleye:~# uptime
 21:44:20 up 22 min,  load average: 0.86, 0.92, 1.03
root@barreleye:~# obmcutil state
 = BMC_READY
root@barreleye:~#

root@barreleye:~# journalctl --no-pager | grep systemd-networkd
Sep 20 21:22:19 barreleye systemd-networkd[678]: Enumeration completed
Sep 20 21:22:22 barreleye systemd-networkd[678]: eth0: Gained carrier
Sep 20 21:22:22 barreleye systemd-networkd[678]: eth0: Gained IPv6LL
``` Sep 20 21:22:36 barreleye systemd-networkd[678]: eth0: Starting DHCPv6 client after NDisc timeout failed: Invalid argument
```
Sep 20 21:22:36 barreleye systemd-networkd[678]: eth0: Configured
Sep 20 21:24:27 barreleye systemd-networkd[678]: eth0: Lost carrier
Sep 20 21:24:27 barreleye systemd-networkd[678]: eth0: Gained carrier
Sep 20 21:24:29 barreleye systemd-networkd[859]: eth0: Gained IPv6LL
Sep 20 21:24:29 barreleye systemd-networkd[859]: Enumeration completed
Sep 20 21:24:47 barreleye systemd-networkd[859]: eth0: Starting DHCPv6 client after NDisc timeout failed: Invalid argument
Sep 20 21:24:47 barreleye systemd-networkd[859]: eth0: Configured

vishwabmc, 2016-10-12:
@gkeishin look at this : https://bbs.archlinux.org/viewtopic.php?id=207597

Fixed in - systemd-228-4 => bug fixed

vishwabmc, 2016-10-12:
paste result of systemctl --version

gkeishin, 2016-10-12:
root@barreleye:~# systemctl --version
systemd 229
+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN
root@barreleye:~#

vishwabmc, 2016-10-12:
okay.. Looks like its still an issue : Suggestion is to use  IPv6AcceptRouterAdvertisements=0 in /etc/systemd/network/00-*.network

gkeishin, 2016-12-05:
@vishwabmc Any update on this ? or please route/suggest to the person who can look into it. Thanks

closed 2017-03-31 by gkeishin