openbmc/openbmc #661: [Barreleye VERSION="v1.99.0-170"]  BMC failed to boot to BMC_READY

gkeishin, 2016-10-12:
The new https://openpower.xyz/job/openbmc-build/298/  build seems broken.
Brad took a look and suggested to open the issue to track this. 

root@barreleye:~# uptime;obmcutil state; cat /etc/os-release
 21:50:54 up 29 min,  load average: 0.38, 0.22, 0.34
 = BMC_STARTING
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-170"
VERSION_ID="v1.99.0-170-gf735c59-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-170"
BUILD_ID="v1.99.0"

Attaching Journal log here -> 
[obmc-journal.txt](https://github.com/openbmc/openbmc/files/524802/obmc-journal.txt)

sivassrr, 2016-10-18:
It will be helpful if we get update on the same.

amboar, 2016-10-24:
At least the hwmon flood should be resolved by the kernel bump committed recently: https://github.com/openbmc/openbmc/issues/600

gkeishin, 2016-11-22:
This is no longer a problem in latter driver.. assuming there is no particular pending here.. closing this one.

closed 2016-11-22 by gkeishin