openbmc/openbmc #662: Witherspoon: Kernel is not getting updated after code update

yadlapati, 2016-10-12:
Copied image-bmc, image-initramfs and image-kernel to /run/initramfs on w5.aus.stglabs.ibm.com. and issued reboot.  Kernel version did not change.

root@witherspoon:~# uname -a
Linux witherspoon 4.7.0-rc6-b172f8b68d85450a12d4348ee4a006ef8844fe95 #1 Tue Sep 13 18:13:34 UTC 2016 armv6l GNU/Linux
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-170"
VERSION_ID="v1.99.0-170-gf735c59-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-170"
BUILD_ID="v1.99.0"

Images are from https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/lastSuccessfulBuild/artifact/images/witherspoon/

From the BMC console, I can see that kernel image is getting updated. Is this a packaging issue with old kernel level in the build?
Updating kernel...
Erasing block: 31/31 (100%) 
Writing kb: 1954/1954 (100%) 
Verifying kb: 1954/1954 (100%)

geissonator, 2016-10-12:
@mdmillerii any thoughts on this one?

mdmillerii, 2016-10-12:
The build date seems to match the full image name from jenkins

	cuImage--4.7+git0+b172f8b68d-r0-aspeed-bmc-opp-witherspoon-20160913180550.dtb	

So it would seem bitbake thought that recipe was up-to-date, or the deploy directory is stale.

shenki, 2016-10-13:
I think the bug is "witherspoon hardcoded to crazy old version of the kernel".

I went to add some new config options to our witherspoon kernel config today and noticed it contained it's own .bb for the kernel that pinned it to a very old version.

https://gerrit.openbmc-project.xyz/761

legoater, 2016-10-14:
I have a recent kernel tree with a couple of patches (mostly Milton's) adding flash support for the ast2500 

https://github.com/legoater/linux/commits/openbmc-4.7

I gave them a good test on a palmetto and witherspoon. But I am still on a 4.7.3 as 4.7.6 has some 
timing issue with the SPI NOR flash. Need to investigate.

legoater, 2016-10-14:
It is running on w1 currently.

shenki, 2016-10-14:
Wrong bug @legoater :) Open a new one for your ast2500 flash. We can update the kernel again once they are merged.

@yadlapati can you please try https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/811/ to see if it resolves your issue?

legoater, 2016-10-14:
@shenki: ok. will do on the kernel component. sorry for the noise.

yadlapati, 2016-10-14:
New build is working with images from https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/811/distro=ubuntu,target=witherspoon/artifact/images/witherspoon:

root@witherspoon:~# uname -a
Linux witherspoon 4.7.7-9fc87f9c1103ad94063f2a819f21117c166e059d #1 Thu Oct 13 23:47:11 UTC 2016 armv6l GNU/Linux

closed 2016-10-14 by williamspatrick