openbmc/openbmc #668: rtc: Document real time clock interfaces in new format

geissonator, 2016-10-13:


geissonator, 2016-11-01:
Reduced from 2 to 1 point because definition of interface is now in #719

williamli80, 2016-12-01:
- use yaml defined in phosphor-dbus-interface to generate sdbusplus binding(new APIs);
- use the new API to replace APIs in phosphor-time-manager repo

williamli80, 2016-12-19:
PR: https://gerrit.openbmc-project.xyz/#/c/1533/

geissonator, 2017-03-24:
https://gerrit.openbmc-project.xyz/#/c/1942/ is the base commit for this

mine260309, 2017-05-19:
With the patch series to be merged, remains tasks:

- [ ] Update `phosphor-time-manager.bb` to use new service name and executable
- [ ] Update `phosphor-host-ipmid/storagehandler.cpp` to use the new service and interface

geissonator, 2017-05-24:
@mine260309 - please also remove the old service so the old application no longer gets put in the flash image?  That may be already implied with your task 1 above.

mine260309, 2017-05-25:
Yup, it should be part of the task 1

rfrandse, 2017-10-18:
<https://gerrit.openbmc-project.xyz/5087> phosphor-time-manager.bb: Use new service name
Resolves  #668 rtc: Implement real time clock interfaces in new format

closed 2017-10-19 by geissonator