openbmc/openbmc #673: [Palmetto VERSION="v1.99.0-172"] BMC MAC Address is changing after reboot

rahulmah, 2016-10-14:
MAC Address is changing after doing reboot from BMC CLI.  Seen this problem with Palmetto at present.

**System was having MAC Address as 6E:B6:4C:EA:24:76**
```
root@palmetto:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 6E:B6:4C:EA:24:76
          inet addr:xx.xx.xx.xx  Bcast:9.3.164.255  Mask:255.255.255.128
          inet6 addr: fe80::6cb6:4cff:feea:2476%984/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:23248 errors:0 dropped:0 overruns:0 frame:0
          TX packets:675 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1330611 (1.2 MiB)  TX bytes:56117 (54.8 KiB)
```
**Trigger Reboot using "reboot" from BMC CLI** 

After reboot, BMC came back to login prompt after 1 mins. But system IP was pingable after around 6 mins. Debuged and found that **MAC address has changed to 6E:B6:4C:EA:24:76 after reboot.**

```
root@palmetto:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 6E:B6:4C:EA:24:76
          inet addr:xx.xx.xx.xx  Bcast:9.3.164.255  Mask:255.255.255.128
          inet6 addr: fe80::6cb6:4cff:feea:2476%984/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:23248 errors:0 dropped:0 overruns:0 frame:0
          TX packets:675 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1330611 (1.2 MiB)  TX bytes:56117 (54.8 KiB)
```
BMC FW info:
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-172"
VERSION_ID="v1.99.0-172-g1807649-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-172"
BUILD_ID="v1.99.0"
root@palmetto:~#
root@palmetto:~# cat /etc/version
20161012145332
root@palmetto:~#

FFDC
[Journal_MAC_address_change.txt](https://github.com/openbmc/openbmc/files/529580/Journal_MAC_address_change.txt)
[dmesg_MAC_Address change.txt](https://github.com/openbmc/openbmc/files/529583/dmesg_MAC_Address.change.txt)

mdmillerii, 2016-10-14:
@rahulmah this says it came back with the same MAC `HWaddr 6E:B6:4C:EA:24:76`

... although the dmesg says otherwise:
```
ftgmac100 1e660000.ethernet: Generated random MAC address 4e:e7:d2:7a:db:cd```

Since we don't have inventory storage for the MAC on palmetto, we need to check if the mac is set in the `u-boot` environment.

Next FFDC needed is: 

1. `fw_printenv`
2.  If `ethaddr` is set to a valid MAC then a serial port log of reboot including `u-boot`

ratagupt, 2016-10-15:
@rahulmah :- did you paste the correct ifconfig output after reboot,I remember the ifconfig was showing different macaddress after reboot..Can you check once more and update the issue?

rahulmah, 2016-10-15:
@ratagupt : Yeah did not pasted it correctly. Corrected now.

rahulmah, 2016-10-17:
@mdmillerii : I have reproduce the issue with same build and captured all the logs which you have asked. This time MAC address was changed from CE:C8:22:C9:BA:A4 to 22:F6:B7:6F:7C:8B after reboot. 

fw_printenv output from the system is as below. Serial console (which I guess has required u-boot data) and other logs from the system are attached below. 

```
root@palmetto:~# fw_printenv
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
bootcmd=bootm 20080000 20300000
bootdelay=2
baudrate=115200
verify=yes
spi_dma=yes
```

[dmesg_MAC_Change.txt](https://github.com/openbmc/openbmc/files/533041/dmesg_MAC_Change.txt)
[Journal_Logs_MAC_Change.txt](https://github.com/openbmc/openbmc/files/533040/Journal_Logs_MAC_Change.txt)
[Serial_Console_Mac_Change.txt](https://github.com/openbmc/openbmc/files/533042/Serial_Console_Mac_Change.txt)

mdmillerii, 2016-10-17:
Your `fw_printenv` output shows the MAC address is not set in `ethaddr`.  Since `palmetto` does not have an inventory item with the MAC address you need to set the mac address either directly in the environment or through REST.

rahulmah, 2016-10-20:
@mdmillerii : Okay. Let me try to set MAC using REST and see if it persist on reboot.

rahulmah, 2016-10-26:
I dont see the issue with latest Palmetto build v1.99.0-190-gc4fb73e.  So closing it.

```
root@palmetto:~# ifconfig | grep HWaddr
eth0      Link encap:Ethernet  HWaddr C2:1A:3A:70:A1:36
root@palmetto:~#
root@palmetto:~# reboot
[  OK  ] Removed slice system-getty.slice.
..
..
root@palmetto:~# ifconfig | grep HWaddr
eth0      Link encap:Ethernet  HWaddr C2:1A:3A:70:A1:36

```

closed 2016-10-26 by rahulmah