openbmc/openbmc #676: [Barreleye VERSION="v1.99.0-173"]  System Network Un reachable when Powered ON

gkeishin, 2016-10-17:
The system went offline from network when powered on.  The system can back online on reboot only.

The system was still able to be accessed from Serial Console and not from other session outside. 

Driver and build details
https://openpower.xyz/job/openbmc-build/300/

root@barreleye:~# cat /etc/version
20161014150717
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-173"
VERSION_ID="v1.99.0-173-g8a844de-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-173"
BUILD_ID="v1.99.0"
root@barreleye:~# obmcutil state
= HOST_BOOTED
root@barreleye:~#

Attached Journald and Console Log for reference.

[barreleye_journald.txt](https://github.com/openbmc/openbmc/files/532719/barreleye_journald.txt)
[obmc-console.log-20161016224000605615_907.txt](https://github.com/openbmc/openbmc/files/532723/obmc-console.log-20161016224000605615_907.txt)

geissonator, 2016-10-17:
George, is this BarPVT?  If so, this is a known issue due to it's enet config and Gavin is working on a fix in https://github.com/openbmc/openbmc/issues/184 (just read the last few entries).

gkeishin, 2016-10-17:
@geissonator Yes that's on BarPVT..  Closing this issue then since https://github.com/openbmc/openbmc/issues/184 is going to address this..

gkeishin, 2016-10-18:
We hit this on again on different system BarPVT   while we were validating the https://gerrit.openbmc-project.xyz/#/c/819/   build https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/848/

root@barreleye:~# cat /etc/version
20161018080832
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-176"
VERSION_ID="v1.99.0-176-ga1607d8-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-176"
BUILD_ID="v1.99.0"
root@barreleye:~#
root@barreleye:~# obmcutil state
= BMC_READY
root@barreleye:~# uptime
21:27:05 up 5 min,  load average: 1.46, 2.53, 1.27
root@barreleye:~#

Powered on ..

root@barreleye:~# obmcutil state
 = HOST_BOOTED 
and the BMC went offline. we can ping the GW from outside the box but not from the BMC..

journal log for this driver attached here --> 
[barreleye_HOST_BOOTED_journald.txt](https://github.com/openbmc/openbmc/files/536818/barreleye_HOST_BOOTED_journald.txt)

gkeishin, 2016-10-18:
@shenki and @gwshan Could you folks take a look at this..

geissonator, 2016-10-18:
@gwshan We'll leave BarD-04 in the fail state for you to look at.  You should have all the access info.  Similar symptom where we lost network when we power on.  I'll ask someone to confirm it has 2 enet cables plugged in.

geissonator, 2016-10-18:
Bashem confirmed it had an eth1 cable plugged in, he unplugged it and everything started working again.  For now we'll just leave it unplugged, we can test your fix on BarPVT once available.

shenki, 2016-10-19:
I logged into BarD-04 and the networking was working fine. Can you clarify what the issue is, and how to reproduce it?

geissonator, 2016-10-19:
Same issue as BarPVT, it had 2 enet cables plugged in, one to eth0 and one to eth1.  During power on, eth0 seems to toggle it's enet connection so it fails over to eth1, which is not setup to use the IP address info from eth0...so we end up with no network.  We fixed BarD-04 today by just unplugging the enet cable to eth1.  I believe @gwshan has a fix coming for this.

gwshan, 2016-10-19:
we should have a fix and it is being pushed upstream. After that, I will
port it to dev-4.7.

shenki, 2016-10-19:
Okay, this is a dupe of https://github.com/openbmc/openbmc/issues/184 as mentioned in comment https://github.com/openbmc/openbmc/issues/676#issuecomment-254193427

Lets track it in there

closed 2016-10-19 by shenki

geissonator, 2016-10-19:
@williamspatrick was concerned we were just kind of dumping all network related issues into #184 .  This is a very specific issue you only hit with eth0 and eth1 cables plugged in.

shenki, 2016-10-19:
@geissonator Yeah, agreed. We will stop re-opening #184 once we land those patches I think