openbmc/openbmc #678: PCA9552 device driver support for both LEDs and GPIOs

spinler, 2016-10-19:
The 9552 chip is connected to both GPIOs and LEDs, so the driver would need to handle both.  The existing leds-pca955x driver only handles LEDs, and I don't believe it handles user selectable blink rates.

spinler, 2016-10-26:
@shenki Can you give your thoughts on this one? If it's possible to add support to the leds-pca955x driver (or maybe create a new driver), to support use of GPIOs and LEDs off of it?

shenki, 2016-11-02:
The current driver does not support exposing the pins as GPIO.

I was looking at the Witherspoon schematic and it appears we do not use any of the 16 pins as LEDs - they are all GPIOs. This would mean we need to write a driver to control the chip as a GPIO multiplexer, but don't need to worry about sharing functionality with LEDs, nor the PWM controls.

Can you confirm that none of the pins are LEDs. If you find they are, please send me an updated schematic :)

In the mean time, lets leave this issue open to track the creation of a new driver for using the PCA8552 as a GPIO controller.

shenki, 2016-11-02:
http://cache.nxp.com/documents/data_sheet/PCA9552.pdf?pspll=1

https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/drivers/leds/leds-pca955x.c

spinler, 2016-11-02:
@shenki The Witherspoon backplane does just use the GPIOs, but there's another 9552 on the Teak (was called Fir in Firestone/Garrison) fan riser card that uses both.  I'll send you the schematics.

shenki, 2016-11-02:
Ok, I see. We will extend the upstream driver to allow using individual lines as GPIOs.

msbarth, 2017-01-23:
I just send an email out to the mailing list in regard to what looks like I'm looking for under this issue.

What is the status of getting a driver for the PCA9552 as a GPIO controller and how would this be structured for access in user space?

shenki, 2017-02-27:
The driver has not been worked on at this stage.

It will show up in the same way as other gpiolib devices; as a sysfs file or chardev.

shenki, 2017-05-04:
@legoater is taking this one on.

As he has pointed out, the work required is very similar to the PCA9532:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/leds/leds-pca9532.c

shenki, 2017-09-25:
This is now upstream

closed 2017-09-25 by shenki