openbmc/openbmc #681: Generate required data for VPD from MRW

geissonator, 2016-10-21:


geissonator, 2016-11-28:
@dkodihal and @ratagupt could you please summarize what you need with this story?

ratagupt, 2016-12-05:
- [ ] Listing of the data which is required  from MRW for HOST VPD.

- [ ] Listing of the data which is required from MRW for openPowerVPD.

- [ ] Listing of the data which is required  from MRW for EEPROM VPD. 

- [ ] Prepare a format of YAML file which is having all the three data. 

- [ ] Parse the data from MRW and generate YAML file including data for HOST VPD.

- [ ] Parse the data from MRW for EEPROM VPD and add it into the YAML file.

- [ ] Parse the data from MRW for openpower VPD and add it into the YAML file.

closed 2017-02-08 by geissonator