openbmc/openbmc #694: Move to upstream IPMI BT driver

shenki, 2016-10-24:
Linux 4.9-rc2 contains our upstream IPMI BT driver.

We should move to this driver, which changes the location of the character device node. Cedric has patches for this.

We should move to matching on the device with udev instead of hardcoding, if that is appropriate.

shenki, 2016-10-24:
- [ ] Revert the old driver and commit the new
- [ ] Update the device tree
- [ ] Move userspace to the new API

The last step should include the bumping of the kernel so we do not break the build.

legoater, 2016-10-24:
you will find a backport of the mainline kernel patches and an update of the device tree here : 

  https://github.com/legoater/linux/commits/openbmc-4.7-ipmi-bt

and an updated version of btbridged for the new kernel : 

https://github.com/legoater/btbridge/commits/master

The ```btbridged``` tree also includes an extension adding a ```--device <FILE>``` command line option which can be used by to start the daemon with a custom device file name, one set up by a udev rule for instance.

adamliyi, 2016-10-26:
Hi @legoater , @shenki asked me to continue work on this issue, will you send your bt driver to mailing list for review? Also the btbridge patch to Gerrit? Or do you want me to do that?

legoater, 2016-10-26:
@adamliyi: the driver is merged in mainline linux. We should  be fine on that side.

I can send the brbridge patches on the mailing list but  I don't think this required if we go through gerrit. 

What I think we would need to do now, is to integrate all these changes in an openbmc patch and maybe add a udev rule to set the bt device name also. Then we can check the whole on gerrit.

adamliyi, 2016-10-27:
Submitted the btbridge patch to Gerrit: https://gerrit.openbmc-project.xyz/#/c/904/
https://gerrit.openbmc-project.xyz/#/c/903/, for review.
Tested on Palmetto, the host CPU boots well.

williamspatrick, 2016-11-01:
@shenki - Is the kernel stuff in already?  Do we need a prereq on a commit with this btbridge stuff I just merged?

shenki, 2016-11-01:
There's been some confusion here. 

This patch should not have been merged: https://gerrit.openbmc-project.xyz/#/c/904/2

I would have preferred to have seen this one merged only: https://gerrit.openbmc-project.xyz/#/c/903/

As I said in the second comment, we needed to move to the new chardev path at the same time as including the new kernel version.

As this was a change that would cause an incompatible break between userspace and the kernel, I was holding off merging it until we could inform everyone and make the change. However, we now have the userspace changed without the kernel bits in.

I will send a patch to bump the kernel version.

shenki, 2016-11-01:
In my pre-coffee confusion, I realise we haven't changed the btbridge version included in openbmc.

@williamspatrick yep, we need to make sure we bump the kernel version in the same commit as bumping the btbridge. I'll do that patch.

shenki, 2016-11-16:
This is complete as of cc1502f3d6c544116ed570b2a02da2cdcdd550c1

closed 2016-11-16 by shenki