openbmc/openbmc #699: Critical issue reported by Nessus security scanner

prkatti1, 2016-10-24:
These are vulnerabilities reported by nessus on openbmc

Severity Plugin Id Name
Critical (10.0) 93650 Dropbear SSH Server < 2016.72 Multiple Vulnerabilities
High (7.8) 80101 IPMI v2.0 Password Hash Disclosure
Medium (6.4) 51192 SSL Certificate Cannot Be Trusted
Medium (6.4) 57582 SSL Self-Signed Certificate
Medium (4.3) 85582 Web Application Potentially Vulnerable to Clickjacking

amboar, 2016-10-26:
@prkatti1 Regarding dropbear, does nessus list the CVEs? Some patches were just applied to staging fixing four: https://patchwork.openembedded.org/patch/133098/

sivassrr, 2016-11-08:
@prkatti1 could you rerun and see whether we are seeing any issues, here

prkatti1, 2016-11-08:
@sivassrr sure, will & share the report.

amboar, 2016-11-14:
The plugin in question is described at http://www.tenable.com/plugins/index.php?view=single&id=93650 and lists the following CVEs:

* CVE-2016-7406
* CVE-2016-7407
* CVE-2016-7408
* CVE-2016-7409

This list lines up with the CVEs mentioned in the patchwork link above, so I'll send a pull request.

williamspatrick, 2016-11-14:
We're going to upgrade to Yocto 2.2 at some point in the not too distant future.  I wouldn't waste time on this issue any more.

williamspatrick, 2016-11-14:
This could also likely be fixed just by picking up Yocto 2.1.1.

amboar, 2016-11-14:
@williamspatrick right; @causten was emailing me about it just now after @prkatti1 sent the report internally.

williamspatrick, 2016-11-14:
Yocto 2.2 has an upgrade of Dropbear to 2016.74, which solves these issues.

None of these CVEs would actually affect us from what I can tell.

williamspatrick, 2017-04-25:
Should be resolved.

closed 2017-04-25 by williamspatrick