openbmc/openbmc #700: [Palmetto- v1.99.0-190-gc4fb73e] : Presence detect field not present for dimm3

rahulmah, 2016-10-24:
In Palmetto, presence detect filed "present" is not available for dimm3. While for other dimm like dimm0, dimm1 and dimm2, it is present. This problem was not seen earlier. It is recently seen with recent build on Palmetto. 

Same problem is not seen on Barelleye system.

**DIMM3 URI output**
```
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx//org/openbmc/inventory/system/chassis/motherboard/dimm3
{
  "data": {
    "Asset Tag": "",
    "Custom Field 1": "",
    "Custom Field 2": "",
    "Custom Field 3": "",
    "Custom Field 4": "",
    "Custom Field 5": "",
    "Custom Field 6": "",
    "Custom Field 7": "",
    "Custom Field 8": "",
    "FRU File ID": "",
    "Manufacturer": "0x9801",
    "Model Number": "9965439-092.A00LF ",
    "Name": "0x0b",
    "Serial Number": "0x1820e291",
    "Version": "0x0000"                  <--- Present field is not there
  },
  "message": "200 OK",
  "status": "ok"
```

**DIMM2 URI output**
```
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx//org/openbmc/inventory/system/chassis/motherboard/dimm2
{
  "data": {
    "Asset Tag": "",
    "Custom Field 1": "",
    "Custom Field 2": "",
    "Custom Field 3": "",
    "Custom Field 4": "",
    "Custom Field 5": "",
    "Custom Field 6": "",
    "Custom Field 7": "",
    "Custom Field 8": "",
    "FRU File ID": "",
    "Manufacturer": "0x9801",
    "Model Number": "9965439-092.A00LF ",
    "Name": "0x0b",
    "Serial Number": "0x1620ec91",
    "Version": "0x0000",
    "fault": "False",
    "fru_type": "DIMM",
    "is_fru": 1,
    "present": "True",           <--- Present field is there
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
}
```

**Openbmc Fw info**
root@palmetto:/tmp# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-190"
VERSION_ID="v1.99.0-190-gc4fb73e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-190"
BUILD_ID="v1.99.0"

root@palmetto:/tmp# cat /etc/version
20161021141723
root@palmetto:/tmp#

**FFDC logs**
[BMC_dmesg_DIMM3_issue.txt](https://github.com/openbmc/openbmc/files/547721/BMC_dmesg_DIMM3_issue.txt)
[BMC_inventory_DIMM3_issue.txt](https://github.com/openbmc/openbmc/files/547723/BMC_inventory_DIMM3_issue.txt)

rahulmah, 2016-10-26:
@spinler : This is a minor problem which is seen with recent builds. Can you please take a look at this?

spinler, 2016-10-27:
Do you guys run this test every day and it just started failing in this driver?

rahulmah, 2016-10-27:
Not running frequently on Palmetto. From my old logs, the last time it was passing was with build v1.99.0-172-g1807649. This build is little old.

rahulmah, 2016-11-03:
With latest openbmc build (v1.99.0-212-gf1c10dd/20161102043616) on Palmetto, I dont see problem with dimm3. But now I see same problem with dimm0. Fields which are missing in dimm0 are fault, fru_type, is_fru, present and version. For all other dimms (like dimm2, dimm1), these are available. So the problem is with dimm0 now.

**DIMM0**
```
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/org/openbmc/inventory/system/chassis/motherboard/dimm0
{
  "data": {
    "Asset Tag": "",
    "Custom Field 1": "",
    "Custom Field 2": "",
    "Custom Field 3": "",
    "Custom Field 4": "",
    "Custom Field 5": "",
    "Custom Field 6": "",
    "Custom Field 7": "",
    "Custom Field 8": "",
    "FRU File ID": "",
    "Manufacturer": "0x9801",
    "Model Number": "9965439-092.A00LF ",
    "Name": "0x0b",
    "Serial Number": "0x1520e691",
    "Version": "0x0000"                     <--- Present, fault, fru_type, is_fru and version field is not there for dimm0
  },
  "message": "200 OK",
  "status": "ok"
```

**DIMM3**
```
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/org/openbmc/inventory/system/chassis/motherboard/dimm3
{
  "data": {
    "Asset Tag": "",
    "Custom Field 1": "",
    "Custom Field 2": "",
    "Custom Field 3": "",
    "Custom Field 4": "",
    "Custom Field 5": "",
    "Custom Field 6": "",
    "Custom Field 7": "",
    "Custom Field 8": "",
    "FRU File ID": "",
    "Manufacturer": "0x9801",
    "Model Number": "9965439-092.A00LF ",
    "Name": "0x0b",
    "Serial Number": "0x1820e291",
    "Version": "0x0000",
    "fault": "False",
    "fru_type": "DIMM",
    "is_fru": 1,
    "present": "True",             <--- Present, fault, fru_type, is_fru and version field is there for dimm3
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
```

gtmills, 2018-05-31:
@rahulmah No update in over a year. Can this be closed? Thanks!

closed 2018-06-01 by rahulmah