openbmc/openbmc #701: [Barreleye build 313] Kernel broken on BARE-01

gkeishin, 2016-10-24:
root@barreleye:~# cat /etc/version
20161024050430

jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00387fa4: 0xc142 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00387fa8: 0xc142 instead
jffs2: Further such events for this erase block will not be printed
jffs2: error: (379) jffs2_link_node_ref: Adding new ref de66ed98 at (0x00388008-0x0038804c) not immediately after previous (0x00388000-0x00388008)
------------[ cut here ]------------
kernel BUG at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/barreleye/openbmc/build/tmp/work-shared/barreleye/kernel-source/fs/jffs2/nodelist.c:622!
Internal error: Oops - BUG: 0 [#1] ARM
CPU: 0 PID: 379 Comm: mount Not tainted 4.7.9-79bcff668f9903e191391979c2d46ebceed0f29f #1
Hardware name: ASpeed SoC
task: de5bc060 ti: dd0b8000 task.ti: dd0b8000
PC is at jffs2_link_node_ref+0xe4/0x188
LR is at jffs2_link_node_ref+0xe4/0x188
pc : [<c022a9d0>]    lr : [<c022a9d0>]    psr: 60000013
sp : dd0b9d28  ip : 60000013  fp : 00000000
r10: 00000000  r9 : c0632130  r8 : 0000e002
r7 : de668cd4  r6 : de57b7a0  r5 : 00000044  r4 : de66ed98
r3 : 3629655d  r2 : 3629655d  r1 : 00000000  r0 : 00000092
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 5e670000  DAC: 00000051
Process mount (pid: 379, stack limit = 0xdd0b8190)
Stack: (0xdd0b9d28 to 0xdd0ba000)
9d20:                   00388008 0038804c 00388000 00388008 de668cf4 de665800
9d40: de57b7a0 c0239b40 de66f560 00000000 0000032c 00000000 00001000 de668cd4
9d60: 20061985 0000032c 00000000 3629655d 00000324 0000032c de665800 00000008
9d80: de57b7a0 de668cd4 00388000 00000000 00000000 c022fad8 c0632130 dd0b9de8
9da0: dd0b9de4 00000000 de668ff8 de668000 00001000 003870dc dd0edb80 dd0b9d98
9dc0: 0000b7a0 de570000 00000000 dd0b8000 00000000 00000388 0000035d dd0edea0
9de0: de502a40 de668000 00000000 00001000 e0021985 00000087 dd0b9e10 3629655d
9e00: de665800 de665800 00000000 de665000 00400000 00008000 c060eb44 00000000
9e20: c02341c4 c02318ec de6651ec c019c0f4 de665000 00000000 00000000 c060200c
9e40: 00000000 3629655d de665000 de665800 00001000 de665000 00400000 00008000
9e60: c060eb44 00000000 c02341c4 c0233c74 de665000 dd172c00 00008000 00000000
9e80: 00008000 c02c8768 dd172c00 00008000 dd09aac0 de003d20 dd09aac0 00000006
9ea0: c060200c c02c8abc dd172c00 c02341c4 de42d980 3629655d 00000000 c0233e58
9ec0: 00008000 c060eb44 c060eb44 00000000 c060eb44 00000000 dd0ed420 c0233e68
9ee0: c02341c4 00008000 dd09aac0 c01a5284 c060eb44 00008000 dd093c20 00008000
9f00: dd09aac0 c01bc934 00000020 00008000 c06070e8 00008000 dd09aac0 c01be3d8
9f20: 00000000 00000000 c060200c 3629655d c060200c bef2bb88 00008000 00000006
9f40: bef2bfbc dd0931b0 de01b330 dd09aac0 0000000f 00000000 dd09aac0 3629655d
9f60: 0000000f bef2bfa8 dd0ed420 dd09aac0 00000000 00008000 dd0b8000 00000000
9f80: 00000000 c01be774 00000000 bef2bfa8 00000000 00008000 b6f6d4d0 00000015
9fa0: c0102344 c01021a0 00000000 00008000 bef2bf99 bef2bfa8 bef2bfbc 00008000
9fc0: 00000000 00008000 b6f6d4d0 00000015 00000000 00000000 00000000 00000000
9fe0: 0002f968 bef2bbd0 00016bcc 480c3e70 60000010 bef2bf99 00000000 00000000
[<c022a9d0>] (jffs2_link_node_ref) from [<c0239b40>] (jffs2_sum_scan_sumnode+0x2ec/0x904)
[<c0239b40>] (jffs2_sum_scan_sumnode) from [<c022fad8>] (jffs2_scan_medium+0x3ec/0x17b8)
[<c022fad8>] (jffs2_scan_medium) from [<c02318ec>] (jffs2_do_mount_fs+0x17c/0x650)
[<c02318ec>] (jffs2_do_mount_fs) from [<c0233c74>] (jffs2_do_fill_super+0x178/0x248)
[<c0233c74>] (jffs2_do_fill_super) from [<c02c8768>] (mount_mtd_aux+0x74/0x100)
[<c02c8768>] (mount_mtd_aux) from [<c02c8abc>] (mount_mtd+0x1b4/0x22c)
[<c02c8abc>] (mount_mtd) from [<c0233e68>] (jffs2_mount+0x10/0x1c)
[<c0233e68>] (jffs2_mount) from [<c01a5284>] (mount_fs+0x10/0x9c)
[<c01a5284>] (mount_fs) from [<c01bc934>] (vfs_kern_mount+0x4c/0xdc)
[<c01bc934>] (vfs_kern_mount) from [<c01be3d8>] (do_mount+0x934/0xab0)
[<c01be3d8>] (do_mount) from [<c01be774>] (SyS_mount+0x70/0x98)
[<c01be774>] (SyS_mount) from [<c01021a0>] (ret_fast_syscall+0x0/0x38)
Code: e59f20a4 e59e1198 e59f00a0 ebfd0424 (e7f001f2)
---[ end trace 27f86c4e40825cf8 ]---
Segmentation fault

Mounting read-write /dev/mtdblock6 filesystem failed.  Please fix and run
        mount /dev/mtdblock6 run/initramfs/rw -t jffs2 -o rw
to to continue, or do change nothing to run from RAM for this boot.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):random: nonblocking pool is initialize

gkeishin, 2016-10-24:
https://openpower.xyz/job/openbmc-build/313/
meta-zaius: Patch U-Boot to enable BMC networking (commit: 266af44) (detail)

geissonator, 2016-10-24:
From Patrick: "We might need to check with @mdmillerii  on it.  There was a patch he had in the older kernel and I wonder if that got dropped with the new NOR support."

mdmillerii, 2016-10-24:
Last time the device tree node wasn't passed to the spi-nor layer and that enabled fast-read in the nor chip but there was no support in the linux driver code, causing a 1 byte shift in all data read.

amboar, 2016-10-25:
I tried to reproduce the issues on a palmetto, so grabbed the palmetto images from build 313. The `magic bitmask` issues appeared, but the rw filesystem mounted successfully (i.e. no segfault occurred in this case).

```
jffs2: Node at 0x00010f84 with length 0x000000ab would run over the end of the erase block
jffs2: Perhaps the file system was created with the wrong erase size?
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00010f88: 0x00ab instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00010f8c: 0xaa73 instead
```

```
root@palmetto:~# mount
dev on /dev type devtmpfs (rw,relatime,size=248224k,nr_inodes=62056,mode=755)
sys on /sys type sysfs (rw,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
/dev/mtdblock5 on /run/initramfs/ro type squashfs (ro,relatime)
/dev/mtdblock6 on /run/initramfs/rw type jffs2 (rw,relatime)
```

amboar, 2016-10-25:
Given:

* `jffs2: Perhaps the file system was created with the wrong erase size?` in the output above
* 313 is built from openbmc/openbmc@266af4458b1b2b3c06d3330f5e20f4aa4aa9fb88,
* which builds a kernel at openbmc/linux@79bcff668f9903e191391979c2d46ebceed0f29f
* which is a patch that apparently [breaks existing jffs filesystems on such flash devices](https://lists.ozlabs.org/pipermail/openbmc/2016-October/005177.html)

I'm building an openbmc image with openbmc/linux@79bcff668f9903e191391979c2d46ebceed0f29f removed and will test.

shenki, 2016-10-25:
I tested https://openpower.xyz/job/openbmc-build/311/ which introduced the new kernel patches on a Witherspoon. I could not reproduce the issue.

@amboar will repeat on a palmetto, in case it's an ast2400 specific issue.

legoater, 2016-10-25:
I suppose the culprit is the 4K erase change on the mx25l25635e : 

 commit 79bcff668f99 ("mtd: spi-nor: add SECT_4K to mx25l25635e")

This is a bad idea as it breaks jff2 and so we should revert the patch I think.

shenki, 2016-10-25:
https://gerrit.openbmc-project.xyz/#/c/892/

closed 2016-10-25 by williamspatrick

gkeishin, 2016-10-25:
https://openpower.xyz/job/openbmc-build/316 came up good on BARPvT