openbmc/openbmc #702: Building qemuarm image on Ubuntu 16.10 fails

mjturek, 2016-10-24:
I have been having trouble building the qemuarm image on Ubuntu 16.10

First I run:
```
. openbmc-env
bitbake obmc-phosphor-image
```
Which eventually fails with the following error:

```
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o eblnonerelocp.o ../../elfutils-0.164/libebl/eblnonerelocp.c
| ../../elfutils-0.164/libebl/eblobjnote.c: In function 'ebl_object_note':
| ../../elfutils-0.164/libebl/eblobjnote.c:45:3: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
|    if (! ebl->object_note (name, type, descsz, desc))
|    ^~
| ../../elfutils-0.164/libebl/eblobjnote.c:135:5: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if'
|      switch (type)
|      ^~~~~~
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o eblrelativerelocp.o ../../elfutils-0.164/libebl/eblrelativerelocp.c
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o eblsysvhashentrysize.o ../../elfutils-0.164/libebl/eblsysvhashentrysize.c
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o eblauxvinfo.o ../../elfutils-0.164/libebl/eblauxvinfo.c
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o eblcheckobjattr.o ../../elfutils-0.164/libebl/eblcheckobjattr.c
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o ebl_check_special_section.o ../../elfutils-0.164/libebl/ebl_check_special_section.c
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o ebl_syscall_abi.o ../../elfutils-0.164/libebl/ebl_syscall_abi.c
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -c -o eblabicfi.o ../../elfutils-0.164/libebl/eblabicfi.c
| cc1: all warnings being treated as errors
| Makefile:538: recipe for target 'eblobjnote.o' failed
| make[2]: *** [eblobjnote.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
| make[2]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/x86_64-linux/elfutils-native/0.164-r0/build/libebl'
| Makefile:473: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/x86_64-linux/elfutils-native/0.164-r0/build'
| Makefile:389: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /home/ubuntu/openbmc/build/tmp/work/x86_64-linux/elfutils-native/0.164-r0/temp/log.do_compile.3215)
ERROR: Task 769 (virtual:native:/home/ubuntu/openbmc/meta/recipes-devtools/elfutils/elfutils_0.164.bb, do_compile) failed with exit code '1'
RAW Paste Data
```

What seems to be happening is that the -Werror flag is set, which causes the elfutils compile to fail due to some pretty benign warnings (bad indentation on the 'if' and 'switch' statements).

I haven't tried any other environments, so it could just be some bad setup on my end.

amboar, 2016-10-24:
Looks like this series should resolve the issue:

https://patchwork.openembedded.org/series/433/

Original pull-request here:

https://patchwork.openembedded.org/patch/127137/

shenki, 2016-10-24:
@mjturek I encourage you to submit those patches to gerrit as backports to our tree

amboar, 2017-01-13:
This is still a problem.

amboar, 2017-01-13:
As a hack I did:

```patch
diff --git a/import-layers/yocto-poky/meta/recipes-devtools/elfutils/elfutils_0.164.bb b/import-layers/yocto-poky/meta/recipes-devtools/elfutils/elfutils_0.164.bb
index 4964c0507854..746ecd09619f 100644
--- a/import-layers/yocto-poky/meta/recipes-devtools/elfutils/elfutils_0.164.bb
+++ b/import-layers/yocto-poky/meta/recipes-devtools/elfutils/elfutils_0.164.bb
@@ -45,6 +45,9 @@ SRC_URI_append_libc-musl = " file://0001-build-Provide-alternatives-for-glibc-as
 
 inherit autotools gettext
 
+CFLAGS_class-native += " -Wno-error=misleading-indentation"
+CFLAGS_class-nativesdk += " -Wno-error=misleading-indentation"
+
 EXTRA_OECONF = "--program-prefix=eu- --without-lzma"
 EXTRA_OECONF_append_class-native = " --without-bzlib"
 EXTRA_OECONF_append_libc-uclibc = " --enable-uclibc"
```

amboar, 2017-01-13:
@mjturek can you please push the patches mentioned above to gerrit?

amboar, 2017-01-13:
Scratch that, I've [pushed the patches myself](https://gerrit.openbmc-project.xyz/#/q/status:open+project:openbmc/openbmc+branch:master+topic:yocto-gcc-native-fixes) because I just don't want to keep running into these problems.

PavamanSubramaniyam, 2017-03-20:
I am also facing this issue on a Ubuntu 17.04 LE OS installed on a P8 Hardware.

| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c -o eblresolvesym.o ../../elfutils-0.164/libebl/eblresolvesym.c
| gcc  -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/share/locale"' -I. -I../../elfutils-0.164/libebl -I..  -I. -I../../elfutils-0.164/libebl -I../../elfutils-0.164/lib -I.. -I../../elfutils-0.164/libebl/../libelf -I../../elfutils-0.164/libebl/../libdw -I../../elfutils-0.164/libebl/../libasm -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Werror -Wunused -Wextra -Wstack-usage=262144  -fPIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c -o eblcheckreloctargettype.o ../../elfutils-0.164/libebl/eblcheckreloctargettype.c
| ../../elfutils-0.164/libebl/eblobjnote.c: In function 'ebl_object_note':
| ../../elfutils-0.164/libebl/eblobjnote.c:45:3: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
|    if (! ebl->object_note (name, type, descsz, desc))
|    ^~
| ../../elfutils-0.164/libebl/eblobjnote.c:135:5: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if'
|      switch (type)
|      ^~~~~~
| cc1: all warnings being treated as errors
| Makefile:538: recipe for target 'eblobjnote.o' failed
| make[2]: *** [eblobjnote.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
| make[2]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/elfutils-native/0.164-r0/build/libebl'
| Makefile:473: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/elfutils-native/0.164-r0/build'
| Makefile:389: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/elfutils-native/0.164-r0/temp/log.do_compile.41187)
ERROR: Task 895 (virtual:native:/home/ubuntu/openbmc/meta/recipes-devtools/elfutils/elfutils_0.164.bb, do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1419 tasks of which 0 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  virtual:native:/home/ubuntu/openbmc/meta/recipes-devtools/elfutils/elfutils_0.164.bb, do_compile
Summary: There were 9 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

PavamanSubramaniyam, 2017-03-20:
The fixes and patches which are pushed here aren't merged yet to the sources.

gtmills, 2018-05-31:
@mjturek No update in over a year. Can this be closed or still a problem? Thanks!

gtmills, 2018-09-04:
@amboar Can we close?

fred0r, 2018-09-18:
This still seems to be a problem.