openbmc/openbmc #704: Barreleye: Do not boot BMC if MAC address is not set instead of setting random MAC

anoo1, 2016-10-25:
@agangidi53 reports that there has been some issues at Barreleye datacenters where systems booting with the same random mac address cause network issues and systems are unreachable.
Therefore this is a request to change the bmc fw logic from setting a random mac when mac is not set, to not letting the BMC boot. Adi will provide further details on the desired behavior.

agangidi53, 2016-10-25:
@anoo1 
**Current Implementation for Barreleye** (Please correct me if I am inaccurate):
1. Scan I/O Board EEPROM for Mac
2. IF Mac address Does not exist in I/O Board EEPROM -> Scan U-Boot env
3. IF Mac address Does not exist in I/O U-boot env ->  Generate Random Mac

**Suggested Implementation for Barreleye** :
1. Scan I/O Board EEPROM for Mac
2. IF Mac address Does not exist in I/O Board EEPROM -> Scan U-Boot env
3. IF Mac address Does not exist in I/O U-boot env ->  Remain in uboot console so that a debug cable can be dropped to enter the unique mac address manually.

agangidi53, 2016-10-31:
Hi @anoo1  any updates ?

gtmills, 2018-06-29:
@anoo1 Can we close?

anoo1, 2018-07-18:
Barreleye is no longer being updated and we didn't get to this implementation. @agangidi53 please reopen if this is still desired on newer platforms.

closed 2018-07-18 by anoo1