openbmc/openbmc #705: Phosphor LED Management Daemon restarting

shenki, 2016-10-25:
Using https://openpower.xyz/job/openbmc-build/311/distro=ubuntu,target=witherspoon/artifact/images/images/images/witherspoon/

Where HEAD is https://github.com/openbmc/openbmc/commit/c4fb73e744d4eb7dc845031dec9aa0543652c02f I see this when booting the BMC:

```
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Started Clear one time boot overrides.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
         Starting Phosphor Inband IPMI...
[  OK  ] Started Serial Getty on ttyS4.
[  OK  ] Reached target Login Prompts.
[  OK  ] Started Avahi mDNS/DNS-SD Stack.
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
```

gkeishin, 2016-11-08:
I also see similar issue on witherspoon as well

root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-226"
VERSION_ID="v1.99.0-226-g068b41c-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-226"
BUILD_ID="v1.99.0"
root@witherspoon:~#

[  OK  ] Started Phosphor Inband IPMI.
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
```[FAILED] Failed to start Phosphor LED Management Daemon.
See 'systemctl status org.openbmc.control.led.service' for details.```

```
● org.openbmc.control.led.service - Phosphor LED Management Daemon
   Loaded: loaded (/lib/systemd/system/org.openbmc.control.led.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Thu 2016-10-20 20:03:20 UTC; 44s ago
  Process: 722 ExecStart=/usr/sbin/led_controller.exe (code=exited, status=0/SUCCESS)
 Main PID: 722 (code=exited, status=0/SUCCESS)
Oct 20 20:03:20 witherspoon systemd[1]: org.openbmc.control.led.service: Service hold-off time over, scheduling restart.
Oct 20 20:03:20 witherspoon systemd[1]: Stopped Phosphor LED Management Daemon.
Oct 20 20:03:20 witherspoon systemd[1]: [[0;1;39morg.openbmc.control.led.service: Start request repeated too quickly.[[0m
Oct 20 20:03:20 witherspoon systemd[1]: [[0;1;31mFailed to start Phosphor LED Management Daemon.[[0m
```

vishwabmc, 2016-11-09:
Is it happening always ?. Has this been fixed in latest ?.

spinler, 2016-11-09:
Witherspoon doesn't have any LEDs defined in the device tree yet.  Is that an error condition in the code?

gkeishin, 2016-11-10:
@vishwabmc I need to see if this shows up in latter.. will report if seen.

vishwabmc, 2016-11-10:
Code is looking at LEDs to be present in /sys/class/leds/ and since nothing is found, its exiting. So this is a generic Witherspoon problem on any build.

root@witherspoon:/usr/sbin# ls /sys/class/leds/
root@witherspoon:/usr/sbin# 

root@witherspoon:/usr/sbin# /usr/sbin/led_controller.exe 
No LEDs present in the system
$

============================

@shenki Something that wanted to ask you about.. when will  the LEDs be populated ?

vishwabmc, 2016-11-10:
@geissonator I don't have any action here. Probably this needs to be closed once @shenki updates on my question.

shenki, 2016-11-10:
@vishwabmc  When someone sends a patch for the device tree to enable them :)

Our userspace should be able to handle missing kernel features like this. It should log a warning that it could not find any leds and then stop trying to re-spawn the daemon. This is why I opened the bug

GHF, 2016-12-16:
If the LEDs in the device tree are populated with the wrong names (or missing the ones it wants to see?), it still fails to start. "Failed to add object to dbus: Invalid argument" is not very useful.

```
Nov 27 19:46:42 zaius systemd[1]: Starting Phosphor LED Management Daemon...
Nov 27 19:46:45 zaius led_controller.exe[762]: Failed to add object to dbus: Invalid argument
Nov 27 19:46:45 zaius led_controller.exe[762]: Error starting LED Services. Exiting
Nov 27 19:46:44 zaius systemd[1]: [[0;1;39morg.openbmc.control.led.service: Main process exited, code=exited, status=234/MAKE_STARTER[[0m
Nov 27 19:46:44 zaius systemd[1]: [[0;1;31mFailed to start Phosphor LED Management Daemon.[[0m
Nov 27 19:46:44 zaius systemd[1]: [[0;1;39morg.openbmc.control.led.service: Unit entered failed state.[[0m
Nov 27 19:46:44 zaius systemd[1]: [[0;1;39morg.openbmc.control.led.service: Failed with result 'exit-code'.[[0m
Nov 27 19:46:45 zaius systemd[1]: org.openbmc.control.led.service: Service hold-off time over, scheduling restart.
Nov 27 19:46:47 zaius led_controller.exe[781]: Failed to add object to dbus: Invalid argument
Nov 27 19:46:45 zaius systemd[1]: Stopped Phosphor LED Management Daemon.
Nov 27 19:46:47 zaius led_controller.exe[781]: Error starting LED Services. Exiting
```

geissonator, 2016-12-16:
@vishwabmc this is all code from skeleton/ledctl/led_controller.c, I think you've pretty much rewrote this all recently?  The code in question seems to be where we're trying to register an object for each LED.
```
497		/* Install the object */
498		rc = sd_bus_add_object_vtable(bus_type,
499				&led_slot,
500				led_object, /* object path */
501				"org.openbmc.Led", /* interface name */
502				led_control_vtable,
503				NULL);
504
505		if(rc < 0)
506		{
507			fprintf(stderr, "Failed to add object to dbus: %s\n", strerror(-rc));
508			break;
509		}
```

vishwabmc, 2017-02-02:
/sys/class/leds is now populated. Can this issue be closed now ?

GHF, 2017-02-07:
I think this is fixed when the daemon can handle missing LEDs in sysfs without crashing.

I still see these crashes on Zaius, and we're not going to be able to work around them without adding e.g. "rear-id" LEDs that the motherboard doesn't have.

vishwabmc, 2017-03-23:
This application will go away sooner since LED is managed differently.

vishwabmc, 2017-03-27:
This can be closed now with https://gerrit.openbmc-project.xyz/#/c/3223/ already merged.

@shenki Please let me know.

vishwabmc, 2017-03-30:
@williamspatrick I suggest we close this and if people have concern, they can reopen ?

closed 2017-03-31 by williamspatrick