openbmc/openbmc #706: Timeout waiting for VUART, fail to reach getty on console

amboar, 2016-10-25:
I've encountered the failure to reach a getty a number of times now on a Palmetto, running openbmc/openbmc@266af4458b1b2b3c06d3330f5e20f4aa4aa9fb88:

```
[FAILED] Failed to start Avahi mDNS/DNS-SD Stack.
See 'systemctl status avahi-daemon.service' for details.
[  OK  ] Started Clear one time boot overrides.
[ TIME ] Timed out waiting for device dev-ttyVUART0.device.
[DEPEND] Dependency failed for Phosphor Cons...tening on device /dev/ttyVUART0.
[FAILED] Failed to start Phosphor LED Management Daemon.
See 'systemctl status org.openbmc.control.led.service' for details.
[FAILED] Failed to start Phosphor Bulk Transfer DBus Bridge.
See 'systemctl status org.openbmc.HostIpmi.service' for details.
[FAILED] Failed to start Phosphor Event Manager.
See 'systemctl status org.openbmc.records.events.service' for details.
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[FAILED] Failed to start Phosphor Download Manager.
See 'systemctl status org.openbmc.managers.Download.service' for details.
[FAILED] Failed to start Phosphor Host0 Checkstop Monitor.
See 'systemctl status org.openbmc.control.Checkstop@0.service' for details.
[FAILED] Failed to start Phosphor Inventory Manager.
See 'systemctl status org.openbmc.Inventory.service' for details.
[FAILED] Failed to start Phosphor Network Manager.
See 'systemctl status org.openbmc.NetworkManager.service' for details.
[FAILED] Failed to start Phosphor Watchdog0.
See 'systemctl status org.openbmc.watchdog.Host@0.service' for details.
[FAILED] Failed to start Phosphor Settings Manager.
See 'systemctl status org.openbmc.settings.Host.service' for details.
[FAILED] Failed to start Phosphor Chassis0 Control.
See 'systemctl status org.openbmc.control.Chassis@0.service' for details.
[FAILED] Failed to start Phosphor Sensor Manager.
See 'systemctl status org.openbmc.Sensors.service' for details.
[FAILED] Failed to start Phosphor User Manager.
See 'systemctl status org.openbmc.UserManager.service' for details.
[FAILED] Failed to start Phosphor BMC0 Control.
See 'systemctl status org.openbmc.control.Bmc@0.service' for details.
[***   ] A start job is running for Phosphor...ery Manager (3min 1s / 5min 50s)random: nonblocking pool is initialized
[ ***  ] A start job is running for Phosphor...ry Manager (4min 13s / 5min 50s)@` `À0`ÀøGot hangup signal
```

Full boot log:

```
U-Boot 2016.07 (Aug 24 2016 - 11:13:50 +0930)

DRAM:  496 MiB
WARNING: Caches not enabled
Flash: 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0
## Booting kernel from Legacy Image at 20080000 ...
   Image Name:   Phosphor OpenBMC (Phosphor OpenB
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1531919 Bytes = 1.5 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 20300000 ...
   Image Name:   obmc-phosphor-image
   Image Type:   ARM Linux RAMDisk Image (lzma compressed)
   Data Size:    1566879 Bytes = 1.5 MiB
   Load Address: 40800000
   Entry Point:  40800000
[... 287 lines cut ...]
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[FAILED] Failed to start Phosphor Download Manager.
See 'systemctl status org.openbmc.managers.Download.service' for details.
[FAILED] Failed to start Phosphor Host0 Checkstop Monitor.
See 'systemctl status org.openbmc.control.Checkstop@0.service' for details.
[FAILED] Failed to start Phosphor Inventory Manager.
See 'systemctl status org.openbmc.Inventory.service' for details.
[FAILED] Failed to start Phosphor Network Manager.
See 'systemctl status org.openbmc.NetworkManager.service' for details.
[FAILED] Failed to start Phosphor Watchdog0.
See 'systemctl status org.openbmc.watchdog.Host@0.service' for details.
[FAILED] Failed to start Phosphor Settings Manager.
See 'systemctl status org.openbmc.settings.Host.service' for details.
[FAILED] Failed to start Phosphor Chassis0 Control.
See 'systemctl status org.openbmc.control.Chassis@0.service' for details.
[FAILED] Failed to start Phosphor Sensor Manager.
See 'systemctl status org.openbmc.Sensors.service' for details.
[FAILED] Failed to start Phosphor User Manager.
See 'systemctl status org.openbmc.UserManager.service' for details.
[FAILED] Failed to start Phosphor BMC0 Control.
See 'systemctl status org.openbmc.control.Bmc@0.service' for details.
[***   ] A start job is running for Phosphor...ery Manager (3min 1s / 5min 50s)random: nonblocking pool is initialized
[ ***  ] A start job is running for Phosphor...ry Manager (4min 13s / 5min 50s)@` `À0`ÀøGot hangup signal
```

amboar, 2016-10-25:
As a note I've just built openbmc/openbmc@385b537ef94c0d73b932d9d57afb9e2e0f08708e, which has booted 3 times in a row without failure.

gkeishin, 2016-10-25:
https://openpower.xyz/job/openbmc-build/316/ we are seeing this on BarPvT..

shenki, 2016-10-25:
@amboar and I took a look a the state of the system from within the initramfs and the kernel driver for the vuart had done the correct thing - we could find both the /dev and the /sys nodes for it.

This is an issue with the systemd unit or something else in userspace

amboar, 2016-10-26:
I couldn't reproduce the issue when bisecting openbmc to find it.

adamliyi, 2016-11-01:
Can you check the ttyVUART0 state when this issue happens:
```
root@palmetto:/# udevadm info -n /dev/ttyVUART0
P: /devices/platform/ahb/ahb:apb/1e787000.vuart/tty/ttyS0
N: ttyS0
S: ttyVUART0
E: DEVLINKS=/dev/ttyVUART0
E: DEVNAME=/dev/ttyS0
E: DEVPATH=/devices/platform/ahb/ahb:apb/1e787000.vuart/tty/ttyS0
E: MAJOR=4
E: MINOR=64
E: SUBSYSTEM=tty
E: TAGS=:systemd:
E: USEC_INITIALIZED=45181752
```
Compared with boot message in normal case, there is a message **Found device /dev/ttyVUART0** from systemd, which is missing from above message when error happens.
```
Oct 27 09:13:23 palmetto kernel[641]: aspeed-smc 1e620000.fmc: Using IRQ 22
Oct 27 09:13:26 palmetto systemd[1]: Found device /dev/ttyS4.
Oct 27 09:13:23 palmetto kernel[641]: aspeed-smc 1e620000.fmc: DMA support deactivated.
Oct 27 09:13:26 palmetto systemd[1]: Found device /dev/ttyVUART0.
Oct 27 09:13:23 palmetto kernel[641]: aspeed-smc 1e620000.fmc: control register changed to: 00000640
Oct 27 09:13:26 palmetto systemd[1]: Started Phosphor Console Muxer listening on device /dev/ttyVUART0.
```

closed 2016-11-07 by williamspatrick