openbmc/openbmc #708: bitbake host-ipmid-tool  is failing

sivassrr, 2016-10-25:
/usr/bin/crossscripts:/tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/sbin:/tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/bin:/tmp/srsiva/build/tmp/sysroots/x86_64-linux/sbin:/tmp/srsiva/build/tmp/sysroots/x86_64-linux/bin:/esw/san2/srsiva/obmc/openbmc/scripts:/esw/san2/srsiva/obmc/openbmc/yocto-poky/bitbake/bin:/esw/san2/iawillia/python/python-2.7.10/bin:/gsa/ausgsa/projects/p/pfd_rat_tools/prod:/usr/lib64/qt-3.3/bin:/usr/local/bin:/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/sbin:/opt/ibm/cmvc/bin:/afs/austin.ibm.com/projects/esw/bin:/usr/ode/bin:/usr/lpp/cmvc/bin"; export HOME="/gsa/ausgsa/home/s/r/srsiva"; git -c core.fsyncobjectfiles=0 read-tree 25a2243e0862ee35f53dc48f471ae61f048ef00e:bin
| DEBUG: Python function base_do_unpack finished
| DEBUG: Python function do_unpack finished
| ERROR: Function failed: Fetcher failure: Fetch command failed with exit code 128, output:
| fatal: Not a valid object name 25a2243e0862ee35f53dc48f471ae61f048ef00e:bin
|
ERROR: Task 94 (/esw/san2/srsiva/obmc/openbmc/meta-phosphor/common/recipes-phosphor/host-ipmid/host-ipmi-hw-example.bb, do_unpack) failed with exit code '1'
NOTE: Tasks Summary: Attempted 419 tasks of which 416 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  /esw/san2/srsiva/obmc/openbmc/meta-phosphor/common/recipes-phosphor/host-ipmid/host-ipmi-hw-example.bb, do_unpack
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
(openbmc)bash-4.1$

sivassrr, 2016-11-02:
Had requested @tomjoseph83 to take a look, He had look at it and initially he found out he couldnt recreate and then he mentioned that his sandbox was pointing to older clone. He will try to check with the latest clone.

tomjoseph83, 2016-11-02:
@sivassrr I just cloned the openbmc project and baked it. I did'nt hit any failures.

williamspatrick, 2016-11-02:
This looks like your download tree has invalid content. I'm not sure what would have caused it to get into that state.

sivassrr, 2016-11-02:
Had did fresh git clone git://github.com/openbmc/ipmitool and did bitbake host-ipmid-tool

Still failed in following way. 

Waiting for 23 running tasks to finish:
ERROR: The recipe cryptodev-linux-native is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /tmp/srsiva/build/tmp/sysroots/x86_64-linux/sysroot-providers/cryptodev-linux-native
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/include/crypto/cryptodev.h
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error however switching DISTRO_FEATURES on an existing build directory is not supported, you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
ERROR: Function failed: sstate_task_postfunc
ERROR: Logfile of failure stored in: /tmp/srsiva/build/tmp/work/x86_64-linux/cryptodev-linux-native/1.7-r0/temp/log.do_populate_sysroot.1846
ERROR: Task 926 (virtual:native:/esw/san2/srsiva/obmc/openbmc/meta/recipes-kernel/cryptodev/cryptodev-linux_1.7.bb, do_populate_sysroot) failed with exit code '1'
ERROR: The recipe gettext-minimal-native is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /tmp/srsiva/build/tmp/sysroots/x86_64-linux/sysroot-providers/gettext-minimal-native
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/po.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/printf-posix.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/glibc2.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/iconv.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/inttypes-pri.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/visibility.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/xsize.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/lib-prefix.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/intl.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/size_max.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/lock.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/lib-ld.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/nls.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/lcmessage.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/extern-inline.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/inttypes_h.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/progtest.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/stdint_h.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/codeset.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/longlong.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/threadlib.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/intldir.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/uintmax_t.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/intlmacosx.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/lib-link.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/gettext.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/intmax.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/intdiv0.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/fcntl-o.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/glibc21.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/wint_t.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/aclocal/wchar_t.m4
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/gettext/config.rpath
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/gettext/po/remove-potcdate.sin
 /tmp/srsiva/build/tmp/sysroots/x86_64-linux/usr/share/gettext/po/Makefile.in.in
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are severa [... cut ...]
NOTE: Tasks Summary: Attempted 206 tasks of which 165 didn't need to be rerun and 3 failed.
Waiting for 0 running tasks to finish:

Summary: 3 tasks failed:
  virtual:native:/esw/san2/srsiva/obmc/openbmc/meta/recipes-devtools/automake/automake_1.15.bb, do_patch
  virtual:native:/esw/san2/srsiva/obmc/openbmc/meta/recipes-kernel/cryptodev/cryptodev-linux_1.7.bb, do_populate_sysroot
  /esw/san2/srsiva/obmc/openbmc/meta/recipes-core/gettext/gettext-minimal-native_0.19.4.bb, do_populate_sysroot
Summary: There were 8 ERROR messages shown, returning a non-zero exit code.

williamli80, 2016-11-08:
After git clone it, get into the ipmitool's path
```
1. . /opt/openbmc-phosphor/2.1/environment-setup-armv5e-openbmc-linux-gnueabi (enable the SDK)
2. ./bootstrap --enable-intf-dbus
3. ./configure --host x86_64
4. make
```
Those we can close this?

mine260309, 2016-11-14:
I just tried to build `ipmi-host-tool` for Romulus without problem, so I think this issue may be closed too.

closed 2016-11-15 by williamspatrick

sivassrr, 2016-11-16:
Thank you Williamli80 , Magic is to remove build directory. It got cleaned up and able to build successfully.