openbmc/openbmc #710: Witherspoon: FSI mux select, gpio326 stuck high in latest code

yadlapati, 2016-10-27:
Paul Robertson is seeing issue using FSP-2 with new OpenBMC code.

root@w5:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-173"
VERSION_ID="v1.99.0-173-g44eacaa-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-173"
BUILD_ID="v1.99.0"
root@w5:~#

His comments from CQ defect (summary: gpio326 is not changing):

The significant gpio difference appears to be gpio326.  It is out,high in the case the of FSI failing, and out,low when it works.  gpio326 is tied to mux select for FSI bus, and should be pulled low automatically when FSI cable is plugged into Sequoia JMFG0.  high would explain why FSI is not linking. gpio326 likely should not be an output in any case, allowing cable to select. ...investigating.

Found that cable attach, does not affect FSI link mux state, just the gpio326. 0 = cronus enabled,  1 = BMC enabled.  This works fine on 4.7.0.* f/w.  However with 4.7.7.* f/w, I'm unable change gpio326 to 0.

root@w9:/sys/class/gpio# echo 0 > gpio326/value
root@w9:/sys/class/gpio# cat gpio326/value
1
root@w9:/sys/class/gpio#

Failure on gpio326 isolated to October f/w.  Reproduced on w9 and w5.  gpio326 is stuck at 1 regardless of output value.

root@w5:~# gpioutil -l A6
326
root@w5:~# gpioutil -p A6 -d out -v 0
root@w5:~# gpioutil -p A6
1                            <<<<<<<< read as 1 even though 0 was written.

This is not a gpioutil problem, as seeing the same issue manipulating gpio directly:

root@w5:~# echo 0 > /sys/class/gpio/gpio326/value
root@w5:~# cat /sys/class/gpio/gpio326/direction
out
root@w5:~# cat /sys/class/gpio/gpio326/value
1

amboar, 2016-10-28:
@yadlapati any chance you or Paul Robertson can dump the pinmux registers on the BMC? 

```shell
for i in 80 84 88 8c 90 94 98 9c a0 a4 a8 ac; do echo 0x1e6e20$i: ` devmem 0x1e6e20$i` done
```

Example output:

```shell
for i in 80 84 88 8c 90 94 98 9c a0 a4 a8 ac; do echo 0x1e6e20$i: ` devmem 0x1e6e20$i` done
0x1e6e2080: 0x00000000
0x1e6e2084: 0x00FFF0C0
0x1e6e2088: 0x01C000FF
0x1e6e208c: 0xC1C000FF
0x1e6e2090: 0x003FA008
0x1e6e2094: 0x00000000
0x1e6e2098: 0x00000000
0x1e6e209c: 0x001FDFF3
0x1e6e20a0: 0x00000000
0x1e6e20a4: 0xFFFF0000
0x1e6e20a8: 0x000FFFFF
0x1e6e20ac: 0x00000000
```

yadlapati, 2016-10-28:
This is from a different system (w5) which is running same level of OpenBMC code

root@w5:~# for i in 80 84 88 8c 90 94 98 9c a0 a4 a8 ac
> do
>  echo 0x1e6e20$i
>  devmem 0x1e6e20$i
> done
0x1e6e2080
0x00000000
0x1e6e2084
0x0000F000
0x1e6e2088
0x03000000
0x1e6e208c
0x00000000
0x1e6e2090
0x0F8F0040
0x1e6e2094
0x00004000
0x1e6e2098
0x00000000
0x1e6e209c
0x023FFFF3
0x1e6e20a0
0x00000000
0x1e6e20a4
0x0000F000
0x1e6e20a8
0x000FFFF0
0x1e6e20ac
0x000000FF

amboar, 2016-10-31:
@yadlapati thanks. It looks like there's at least one bug in the pinmux driver for the ast2500. Bit SCU90[6] is being set, but the datasheet says "Reserved, must keep at value ”0”". As an experiment, can you see if this fixes your issue:

```
# devmem 0x1e6e2090 32 0x0F8F0000
```

Regardless, I'll send a patch fixing that bug.

shenki, 2016-10-31:
Fix is at http://patchwork.ozlabs.org/patch/689212/

I will integrate this into the new kernel revision.

paulgrobertson, 2016-10-31:
That seems to solved the issue.

root@w5:~# gpioutil -p A6 -d out -v 0
root@w5:~# gpioutil -p A6
1

root@w5:~# devmem 0x1e6e2090 32 0x0F8F0000
root@w5:~# gpioutil -p A6
0

root@w5:~# gpioutil -p A6 -d out -v 1
root@w5:~# gpioutil -p A6
1

root@w5:~# gpioutil -p A6 -d out -v 0
root@w5:~# gpioutil -p A6
0

amboar, 2016-10-31:
@paulgrobertson thanks for testing. @shenki is planning to bump the openbmc kernel today with the patch linked above.

williamspatrick, 2016-11-01:
Merged fix via 913e81dbf0910f931497fc5b93fedf9fa39f12fc

closed 2016-11-01 by williamspatrick