openbmc/openbmc #713: pflash/flasher hang blocks boot

GHF, 2016-10-29:
This is a problem I saw on Zaius BMC but I don't think it's specific to us. It looks like pflash is called by op-inventory-upload without `-m` (MTD access), causing it to busy wait and block the boot. Eventually pflash is killed, causing the boot to fail altogether:

```
Oct 21 16:43:09 zaius systemd[1]: op-inventory-upload@0.service: Start operation timed out. Terminating.
Oct 21 16:43:09 zaius systemd[1]: Failed to start op-inventory-upload@0.service.
Oct 21 16:43:09 zaius systemd[1]: op-inventory-upload@0.service: Unit entered failed state.
Oct 21 16:43:09 zaius systemd[1]: op-inventory-upload@0.service: Failed with result 'timeout'.
```

I think there's a similar busy wait from flasher.exe instances spawned by org.openbmc.control.Flash at boot. These run in the background when the BMC comes up and eat cycles for a good long while.

williamspatrick, 2016-10-29:
@shenki - I believe there was a change to pflash coming to resolve this, correct?  Do you know where that is?  I think Stewart has some rc tags we could point at.

shenki, 2016-10-30:
yeah, this could be resolved by bumping to the skiboot-5.4.0-rc2 tag

shenki, 2016-10-31:
https://gerrit.openbmc-project.xyz/#/c/907/

GHF, 2016-11-01:
Great, we also need that version bump to pick up this fix: https://github.com/open-power/skiboot/commit/9c3b35a15f3a2762288e1e460843c9c1246e1acd

Otherwise pflash will always fail with "Architecture doesn't support write protection on flash" when erasing/writing to host PNOR.

shenki, 2016-11-01:
We're blocked on #714 which breaks flasher. I'm reworking flasher to resolve this

closed 2016-11-04 by williamspatrick