openbmc/openbmc #718: [Barrelleye VERSION_ID="v1.99.0-200-g48d0241"] dev-ttyVUART0.device start timed out.

gkeishin, 2016-10-31:
**root@barreleye:~# cat /etc/version
20161031164543
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-200"
VERSION_ID="v1.99.0-200-g48d0241-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-200"
BUILD_ID="v1.99.0"
root@barreleye:~# obmcutil state
 = BMC_READY
root@barreleye:~#

**Journal snippet** 
```
Oct 20 19:58:52 barreleye systemd[1]: dev-ttyVUART0.device: Job dev-ttyVUART0.device/start timed out.
Oct 20 19:58:52 barreleye systemd[1]: Timed out waiting for device dev-ttyVUART0.device.
Oct 20 19:58:52 barreleye systemd[1]: Dependency failed for Phosphor Console Muxer listening on device /dev/ttyVUART0.
Oct 20 19:58:52 barreleye systemd[1]: obmc-console@ttyVUART0.service: Job obmc-console@ttyVUART0.service/start failed with result 'dependency'.
Oct 20 19:58:52 barreleye systemd[1]: dev-ttyVUART0.device: Job dev-ttyVUART0.device/start failed with result 'timeout'.
```

gkeishin, 2016-10-31:
Attached syslog for reference  
[20161031135405219834_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/562764/20161031135405219834_BMC_journalctl.txt)

geissonator, 2016-11-01:
@adamliyi - You're the pro in this area now :)  Any thoughts?

adamliyi, 2016-11-01:
Related issue: https://github.com/openbmc/openbmc/issues/706
Can you check the ttyVUART0 state when this issue happens:
```
root@palmetto:/# udevadm info -n /dev/ttyVUART0
P: /devices/platform/ahb/ahb:apb/1e787000.vuart/tty/ttyS0
N: ttyS0
S: ttyVUART0
E: DEVLINKS=/dev/ttyVUART0
E: DEVNAME=/dev/ttyS0
E: DEVPATH=/devices/platform/ahb/ahb:apb/1e787000.vuart/tty/ttyS0
E: MAJOR=4
E: MINOR=64
E: SUBSYSTEM=tty
E: TAGS=:systemd:
E: USEC_INITIALIZED=45181752
```
Also:
```
systemctl status dev-ttyVUART0.device --no-pager
systemctl start dev-ttyVUART0.device
```

gkeishin, 2016-11-01:
root@barreleye:~# udevadm info -n /dev/ttyVUART0
device node not found
root@barreleye:~# 

root@barreleye:~# systemctl status dev-ttyVUART0.device --no-pager
● dev-ttyVUART0.device
   Loaded: loaded
   Active: inactive (dead)

Oct 20 19:58:53 barreleye systemd[1]: dev-ttyVUART0.device: Job dev-ttyVUART0.device/start timed out.
Oct 20 19:58:53 barreleye systemd[1]: Timed out waiting for device dev-ttyVUART0.device.
Oct 20 19:58:53 barreleye systemd[1]: dev-ttyVUART0.device: Job dev-ttyVUART0.device/start failed with result 'timeout'.
root@barreleye:~#

root@barreleye:~# systemctl start dev-ttyVUART0.device
Job for dev-ttyVUART0.device timed out.
root@barreleye:~#

adamliyi, 2016-11-02:
Thanks @gkeishin .
It looks this issue does not happen always, is there a Barreleye with this bug, so I can login in to debug?
Could you also check whether `/dev/ttyS0` and `/sys/devices/platform/ahb/ahb:apb/1e787000.vuart/tty/ttyS0` exist?
According to @shenki 's comments in openbmc/openbmc#706, these devices nodes are existing, if so we can narrow down this problem to udev/systemd - for some reason udev failed to create /dev/ttyVUART0.
Also if this issue does not happen all the time, I'd suggest to mark it as low priority.

closed 2016-11-07 by williamspatrick