openbmc/openbmc #723: Run CI images under Aspeed Qemu model

shenki, 2016-11-01:
The following machines should boot on openbmc/qemu master:

 * Witherspoon `-M witherspoon-bmc`
 * Palmetto `-M palmetto-bmc`
 * ast2500-evb `-M ast2500-evb`

We should take the artefacts from these CI builds and boot them

@geissonator may already have an issue for this

geissonator, 2016-11-28:
Should just need a tweak to https://github.com/openbmc/openbmc-build-scripts/blob/master/scripts/boot-qemu.sh to support an optional machine param and then set it up in the main script - https://github.com/openbmc/openbmc-build-scripts/blob/master/run-qemu-robot-test.sh 
Then just a change to the jenkins job to call it appropriately.  Unfortunately though I've been chasing an intermittent CI failure for quite some time on openpower.xyz (example https://openpower.xyz/job/openbmc-test-qemu-ci/926/console).  I need to spend some time looking into it.

geissonator, 2017-10-13:
#2272 and #1057 are similar

amboar, 2017-10-16:
We now also have `-M witherspoon-bmc`.

geissonator, 2018-01-17:
I'm working on this in #2272 so just going to close this out since I have more details in that one.

closed 2018-01-17 by geissonator